PX4/PX4-Avoidance #82: local_planner_stereo.launch problem: ODOM: invalid body frame orientation "frd"

ArkadiuszNiemiec, 2018-06-11:
I have made `avoidance` simulation work on the Ubuntu 18.04 (everything worked fine but I had to build `octomap_mapping` and `octomap_ros`). 

The only problem is that I cannot make `local_planner_stereo.launch` to start, this may be a problem:
```
[FATAL] []: ODOM: invalid body frame orientation "frd"
[ WARN] []: MSG to TF: Quaternion Not Properly Normalized
[ INFO] []:  Pointcloud timeout: Hovering at current position 
```

and 

The strange part is that `local_planner_depth-camera.lauch` and `global_planner` work perfectly. Do you know what's the problem?

Whole log: [gist](https://gist.github.com/ArkadiuszNiemiec/224f866aacc5b6e530e6e98c16d54dff)

ArkadiuszNiemiec, 2018-06-11:
```
[FATAL] []: ODOM: invalid body frame orientation "frd"
```
also shows in `global_planner` which works, so it cannot be the source of the problem.

TSC21, 2018-06-11:
That's a problem on MAVROS package that was solved on the latest release. You will have to wait for latest release but if you don't want to wait, you can just clone MAVROS to your workspace and build it from source.

mrivi, 2018-06-11:
@ArkadiuszNiemiec the problem is that you're not getting the point cloud. I am checking what's going on

mrivi, 2018-06-11:
@ArkadiuszNiemiec if you change the parameter `callback_topic` to `/stereo/points2` in rqt reconfigure under local_planner_node it should work. 

The problem is that the parameter in `local_planner_stereo.launch` to set the correct point cloud topic is being overwritten by the callback to rqt reconfigure. I'll make a PR to fix it.

ArkadiuszNiemiec, 2018-06-11:
@TSC21 It worked, no more errors, thank you,
@mrivi changing the topic fixed the problem, thank you! I am waiting for you PR to be merged :smiley:

closed 2018-06-11 by ArkadiuszNiemiec