PX4/PX4-Avoidance #85: problem with docker git

asimonov, 2018-06-12:
submodule update fails for me in docker for libuavcan with 
```
Unable to checkout 'e1c9a4f5064a336295125da25c0542a5cd601dd7' in submodule path 'src/modules/uavcan/libuavcan'
```

looks like git version issue.

if I add this 

```
# Install latest git to avoid 
# `Unable to checkout 'e1c9a4f5064a336295125da25c0542a5cd601dd7' in submodule path 'src/modules/uavcan/libuavcan'`
RUN apt-get install -y software-properties-common && \
    add-apt-repository ppa:git-core/ppa && \
    apt-get update && \
    apt-get install -y git
```

into dockerfile before checking out Firmware it works.

JonasVautherin, 2018-06-12:
Which docker image do you run? Do you have a Dockerfile to share, so that we can reproduce?

asimonov, 2018-06-13:
I have run the standard one provided in https://github.com/PX4/avoidance/tree/master/docker/demo <https://github.com/PX4/avoidance/tree/master/docker/demo>
Think standard ubuntu git is 2.7 which did not work for me in that container.

JonasVautherin, 2018-06-13:
I could build it with `docker-compose build --no-cache` just now :thinking:... Why do you say it is a git issue?

asimonov, 2018-06-13:
> I could build it with docker-compose build --no-cache just now 🤔... Why do you say it is a git issue?

Because when i was running it the git inside the container (standard ubuntu 16.04 version of git) was giving me the error i posted. Using custom (latest) git inside container solved it for me

_____________________________
Kind Regards,
Alexey Simonov

JonasVautherin, 2018-06-14:
Right. I'm not sure it means that the git version should be updated. Maybe it is something else.

Would you mind trying to build the image again (without your modification), i.e. with `$ docker-compose build --no-cache`?

baumanta, 2019-04-09:
seems to be stale

closed 2019-04-09 by baumanta