PX4/PX4-Avoidance #86: Docker demo fails on apt-key adv

Jaeyoung-Lim, 2018-06-13:
docker demo fails to build using `docker-compose up`
```
 ---> Running in fe0c65ad0217
Executing: /tmp/tmp.qUUyo5HYwi/gpg.1.sh --keyserver
hkp://ha.pool.sks-keyservers.net:80
--recv-key
421C365BD9FF1F717815A3895523BAEEB01FA116
gpg: requesting key B01FA116 from hkp server ha.pool.sks-keyservers.net
gpg: keyserver timed out
gpg: keyserver receive failed: keyserver error
ERROR: Service 'ubuntu-avoidance' failed to build: The command '/bin/sh -c echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list &&     apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update' returned a non-zero code: 2
```

JonasVautherin, 2018-06-13:
Do you have an idea why this is happening? I won't have time to check that in the next few days, unfortunately :-/.

Jaeyoung-Lim, 2018-06-13:
@JonasVautherin I will look more into it

JonasVautherin, 2018-06-13:
BTW I tried to build again for [this issue](https://github.com/PX4/avoidance/issues/85), and it worked for me.

I was running the following command from `docker/demo/ubuntu`:

```
docker-compose build --no-cache
```

JonasVautherin, 2018-06-13:
Just ran it on another computer (macOS, this time), and worked again:

``` ---> Running in 9542ea0913ce
Executing: /tmp/tmp.q4H5D9lzpL/gpg.1.sh --keyserver
hkp://ha.pool.sks-keyservers.net:80
--recv-key
421C365BD9FF1F717815A3895523BAEEB01FA116
gpg: requesting key B01FA116 from hkp server ha.pool.sks-keyservers.net
gpg: key B01FA116: public key "ROS Builder <rosbuild@ros.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1```

shairubin, 2018-07-06:
mavros build seems to hang on Mac 
using **'docker-compose up'** 

[build] Found '6' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting >>> catkin_tools_prebuild                                             
Finished <<< catkin_tools_prebuild                [ 1.9 seconds ]              
Starting >>> mavlink                                                           
Finished <<< mavlink                              [ 4.0 seconds ]              
Starting >>> mavros_msgs                                                       
Finished <<< mavros_msgs                          [ 10.0 seconds ]             
Starting >>> libmavconn                                                        
Finished <<< libmavconn                           [ 7.5 seconds ]              
Starting >>> mavros                                                            
^[[Ald 14:52.0 s] [4/7 complete] [4/4 jobs] [0 queued] [High Load] [mavros:... 
**[build 14:59.1 s] [4/7 complete] [4/4 jobs] [0 queued] [High Load] [mavros:...**

JonasVautherin, 2018-07-06:
I would try to increase the memory allocated to the docker VM on macOS. Because there it is virtualized, so you get lower performances. Not sure if that will help, though =/.

closed 2019-03-27 by Jaeyoung-Lim