PX4/PX4-Avoidance #87: docker-compose installation instructions should be updated

Jaeyoung-Lim, 2018-06-13:
According to the [readme](https://github.com/PX4/avoidance/tree/master/docker/demo) in the docker folder, the docker-compose installation is instructed to be done as
```
apt install docker-compose
```

However, the docker-compose installed using apt is `1.8.1` which is outdated. 
This produces an error as the `docker-compose.yaml` uses `version: '3'` which is [not compatible](https://docs.docker.com/compose/compose-file/compose-versioning/) with `1.8.1`

A better approach is to install the docker-compose using the instructions provided [here](https://docs.docker.com/compose/install/)

JonasVautherin, 2018-06-13:
That's what I was doing [here](https://github.com/PX4/avoidance/blob/master/docker/run.py#L70), actually. Would be worth updating the docker-compose version there as well.

shairubin, 2018-07-04:
when trying to run the run.py I get

File "run.py", line 117, in <module>
    main(args)
  File "run.py", line 10, in main
    runGlobalPrerequisites()
  File "run.py", line 54, in runGlobalPrerequisites
    while isDockerComposeNotInstalled():
  File "run.py", line 58, in isDockerComposeNotInstalled
    return shutil.which("docker-compose") == ''
AttributeError: 'module' object has no attribute 'which'

JonasVautherin, 2018-07-05:
@shairubin: Are you running `run.py` with python3?

shairubin, 2018-07-05:
No — will try

closed 2019-03-28 by Jaeyoung-Lim