PX4/PX4-Avoidance #91: Link to instructions to run on odroid is broken

mzahana, 2018-07-13:
Hi.

The link that points to running the package on odroid is broken.
https://github.com/PX4/avoidance#running-on-odroid

Could you please update it?

Thanks.

adipandas, 2018-07-23:
Updated link: [Running on Odroid](https://github.com/PX4/avoidance/tree/master/global_planner/resource/odroid)

mzahana, 2018-07-23:
Thanks. Which planner is tested to work outdoor? I see the README under the global planner.
Also, I have an Intel Up board with Realsense R200. Would the setup on the Up board be similar?
Thanks again.

mrivi, 2018-07-23:
@mzahana Global planner was tested outdoor, running on odroid.
However, right now most of our efforts are on the local planner. We are running it on Intel Aero (https://docs.px4.io/en/flight_controller/intel_aero.html)

closed 2018-07-23 by mrivi

mzahana, 2018-07-23:
Can the same setup on odroid be used to run local planer?

mzahana, 2018-07-23:
**Notes**:
* There is NO `odroid` branch in [PX4/avoidance](https://github.com/PX4/avoidance)
* There is NO `half_resolution` branch in [PX4/uvc_ros_driver](https://github.com/PX4/uvc_ros_driver)

PR #93 (2018-07-23): Fix Odroid link in README
Fixes #91.
commit ace61a91 (2018-07-23): Fix Odroid link in README

Fixes #91.