PX4/PX4-Avoidance #94: R200 parameters for outdoor

mzahana, 2018-07-24:
Hi I am currently testing the avoidance pkg in an outdoor setup. I am using the R200 dvkit.  I suspect that the camera is not tuned for outdoor operations. Do you have recommended R200 parameters for outdoor operations?
I am using the global planet in the master branch. It always reports that the goal position is occupied while it is actually not.

Thanks

baumanta, 2018-09-06:
when using the realsense cameras, it often is an issue that the auto exposure is not turned on or not working by default. Try to disable and enable the auto exposure using rqt reconfigure. You can check whether the camera is working correctly by looking at the images and pointcloud using Rviz.

The global planner has a few known issues. Maybe what you are experiencing is connected to #33

aurashn, 2018-11-05:
@mzahana we’re you able to use the r200 with the up board for avoidance?

aurashn, 2018-11-14:
@mzahana we are also going to use r200 in outdoor environment, with px4 firmware on our pixhawk 2.1 cube, did changing the auto exposure fix this issue for you? Also, could you list a few key steps to getting the r200 working with actual hardware. I would really appreciate this!

closed 2018-12-14 by baumanta