PX4/PX4-Avoidance #97: Check usage of std::clock()

bkueng, 2018-08-14:
[std::clock](https://en.cppreference.com/w/cpp/chrono/c/clock) measures the processor time and is in many cases not what you want.

For example it's used to calculate `dt` in https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/waypoint_generator.cpp#L243.
This is not correct and [std::chrono::high_resolution_clock::now](https://en.cppreference.com/w/cpp/chrono/high_resolution_clock/now) should be used instead.

`std::clock` is used at several places and all of them need to be validated.

baumanta, 2018-08-15:
Right, thanks for looking into it! I agree, this is wrong. I used ros::Time:now() in some cases. For constistency, I think we should choose either that or chrono. Which on do you like better?

http://wiki.ros.org/roscpp/Overview/Time

baumanta, 2018-08-15:
Thanks! I think I''ll go with ros::Time::now(). Just to keep all options open for simulation

nicovanduijn, 2018-08-24:
Has been taken care of as part of #102 (7be4b85)

closed 2018-08-24 by nicovanduijn