PX4/PX4-Avoidance #104: local_planner_stereo.launch fails to run

ChuplesKai, 2018-09-03:
Directly attempting to run this launch file resulted in an error in looking up the px4 launch file, which makes sense because there is no px4.launch file in the PX4 firmware folder.

So I figure the launch files are probably just a little out of date and I need to update some directories.  I change it to look for px4.launch in mavros, where it does find it, but it then complains about multiple instances of mavros running.  Indeed, px4.launch does start its own mavros node, so I tried to consolidate all of the launch files into a single file to try running the software 
[local_planner_stereo.txt](https://github.com/PX4/avoidance/files/2346630/local_planner_stereo.txt).

I believe this has the correct configuration, but I now run into strange errors from the parser:

`Error [parser.cc:689] Unable to find uri[model://iris_depth_camera]
<sdf version='1.6'>
  <world name='default'>
    <gravity>0 0 -9.8</gravity>
    <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
    <atmosphere type='adiabatic'/>
    <physics name='default_physics' default='0' type='ode'>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
    </physics>
    <scene>
      <ambient>0.4 0.4 0.4 1</ambient>
      <background>0.7 0.7 0.7 1</background>
      <shadows>1</shadows>
    </scene>
  </world>
</sdf>
[INFO] [1536011813.196405, 10.104000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris
[spawn_model-11] process has finished cleanly
log file: /home/chuples/.ros/log/1c59b90e-afc4-11e8-a497-dcfe07d69cab/spawn_model-11*.log
^C[gazebo_gui-10] killing on exit
[gazebo-9] killing on exit
[local_planner_node-6] killing on exit
[rqt_reconfigure-7] killing on exit
[rviz-4] killing on exit
[mavros-8] killing on exit
[stereo/stereo_image_proc-5] killing on exit
[tf_90_deg-2] killing on exit
[tf_depth_camera-3] killing on exit
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
[local_planner_node-6] escalating to SIGTERM
[gazebo-9] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
`

bkueng, 2018-09-04:
Please update the PX4Firmware to latest master, the px4.launch file will be there.

closed 2018-09-04 by bkueng

mhkabir, 2018-09-04:
You probably didn't follow the readme setup instructions to add the avoidance repo to GAZEBO_MODEL_PATH.

mhkabir, 2018-09-04:
```

ChuplesKai, 2018-09-04:
Okay, thanks mhkabir, I did miss those steps.  Everything seems to run now.