PX4/PX4-Avoidance #115: Documentation

sarahToonsi, 2018-10-07:
Hi 
I want to use this package with different cameras (other than realsense) and I want to be able to have full control on parameters and topics, however I cant find proper documnetation about the nature of the parameters and topics. 
Can someone help :)

nicovanduijn, 2018-10-09:
Heads up: This will mean quite some integration work to make it work.

What type of cameras are you looking to use? You will need some sort of 3D information, so at the very least a stereo rig will be required. Then I would suggest looking into the [ROS stereo image processing tools](http://wiki.ros.org/stereo_image_proc).

And you're right in that the parameters are not yet very well documented. Feel free to contribute;)

mrivi, 2019-02-18:
Documentation has improved since this issue was open

closed 2019-02-18 by mrivi