PX4/PX4-Avoidance #117: OA: in manual mode ("collision avoidance")

AuterionWrikeBot, 2018-10-15:
**Goal**: in manual mode obstacle avoidance is used to avoid crashing into obstacles.
**Intended behavior:** not full avoidance of obstacle is needed, main goal is to avoid a crash.

* Ideas:
* 1) just stop when flight path into obstacle is detected. Might feel bad to the user because drone does not move even though he provides stick input
* 2) bounce back (idea by Jonathan): when you push sticks forward into an obstacle bounce back slightly (like the browser on an iphone when you scroll further up at the top of a page)

**Use case**: inspection

AuterionWrikeBot, 2018-10-15:
➤ Wrike Bot commented:

@Tanja Baumann All subtasks have been completed or deferred, you might want to complete this task as well.

thomasgubler, 2018-10-15:
@bresch @mhkabir FYI

mhkabir, 2018-10-15:
Some initial work is here : https://github.com/mhkabir/Firmware/tree/pr-obstacle-avoidance

Tested in simulation with single forward rangefinder, pending testing on real vehicle.

baumanta, 2018-10-17:
I opened up PR  [#10710](https://github.com/PX4/Firmware/pull/10710)  for discussion. In SITL the behavior looks reasonable, but might leave room for improvements. I'm open for inputs