PX4/PX4-Avoidance #126: Unable to find PX4

rajat2004, 2018-10-28:
I'm getting the following error while trying to launch the simulation.
`$ roslaunch global_planner global_planner_sitl_mavros.launch`
```
... logging to /home/rajat/.ros/log/5e6ec4d2-daa4-11e8-aa40-4cbb58b223ff/roslaunch-rajat-Vostro-3449-6976.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: px4
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/rajat/catkin_ws/src/avoidance/global_planner
ROS path [2]=/home/rajat/catkin_ws/src/avoidance/local_planner
ROS path [3]=/opt/ros/kinetic/share

```
The simulation works when I run `make posix_sitl_default gazebo`

rajat2004, 2018-10-30:
Problem solved
This was caused as I was setting `ROS_PACKAGE_PATH` before sourcing `~/catkin_ws/devel/setup.bash` which removed `~/Firmware` from the path. Exporting the paths after sourcing the `setup.bash` file solves the problem.

closed 2018-10-30 by rajat2004

yc718764666, 2018-11-22:
Hi, I have the same question with you, could you explain it in more detail? thank you very much.

rajat2004, 2018-11-22:
Sorry, my previous answer wasn't clear enough.

Before running the simulation, you need to source the `setup.bash` script in the catkin workspace using `source ~/catkin_ws/devel/setup.bash`.

This removed `~/Firmware` from `ROS_PACKAGE_PATH` which was exported in the previous step.
Due to this, ROS was unable to find the `px4.launch` file located in `~/Firmware/launch`

So after having sourced the `setup.bash` script, just export `ROS_PACKAGE_PATH` again using
`export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware` .

I added the commands in the following order in my `~/.zshrc` to automatically source the scripts and export the paths everytime the terminal is opened.
```
source ~/catkin_ws/devel/setup.zsh
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/posix_sitl_default
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware
```

If you are using `bash`, just replace `zsh` with `bash`

yc718764666, 2018-11-22:
Thank you very much for your reply. I have already solved this kind of problem.