PX4/PX4-Avoidance #128: Simulation

HasanGilani, 2018-10-31:
I am trying to run local planner on iris drone with r200 camera. I have changed the point cloud topic and it shows it in rviz but it does not move to the goal. If I give the goal through rviz, the goal's coordinates are shown on terminal but drone doesn't move

mrivi, 2018-10-31:
@HasanGilani you need to arm the vehicle and switch to offboard mode. You can do it in QGroundControl or by running the following commands in the mavlink shell
```rosrun mavros mavsys mode -c OFFBOARD
  rosrun mavros mavsafety arm ```

HasanGilani, 2018-11-01:
These commands don't work but even before i was arming the drone and switching to offboard mode. I can also control the drone by publishing to different topics. Bus this package is not able to move the drone.

HasanGilani, 2018-11-01:
@mrivi I  am using local_planner_stereo.launch and getting this warning:
Point cloud timeout, missing transforms  (Hovering at current position).
I have replaced the topic in launch file with my point clouds topic.

mrivi, 2018-11-01:
@HasanGilani are you using the latest master? Could you please try to use the latest master with the default vehicle model? I have just tried that and offboard works for me. 
If that is working there is something wrong with the modification you have made.
Can you please copy paste here the entire output of the avoidance together with `rostopic hz ` on your point cloud topic and on `/mavros/local_position/pose` and the modified launch file. Thanks

closed 2019-02-18 by mrivi