PX4/PX4-Avoidance #132: [Feature] Plug OMPL to Avoidance to open the possibility of using other motion planning algorithms

TSC21, 2018-11-02:
[OMPL](http://ompl.kavrakilab.org/) (Open Motion Planning Library) is a library which contains most of the state-of-the-art motion and path planning algorithms. It's already used in various robotics frameworks, including MoveIt!. Would be interesting to explore the possibility of integrate it with the Avoidance API in order to add other alternatives the global planner.
Note also that the [license](http://ompl.kavrakilab.org/license.html) is permissive and should set us good to go on the integration.
I am open to discuss this on the next avoidance call.
@mrivi, @baumanta, @nicovanduijn what do you think?

TSC21, 2018-11-02:
Note that it can also be installed directly from the ROS deb packages using `sudo apt-get install ros-`rosversion -d`-ompl`.

Jaeyoung-Lim, 2018-11-20:
I would be excited to see this moving forward

TSC21, 2018-11-20:
@Jaeyoung-Lim is this something you have the bandwidth to explore?

ytaskiran, 2021-04-29:
What happened to this project? It seemed some progress had taken place at the time but I don't understand where it went.

Jaeyoung-Lim, 2021-04-29:
@ytaskiran This is how far it went: https://github.com/PX4/PX4-Avoidance/pull/453

LorenzMeier, 2021-04-29:
@ytaskiran Would you want to take it on from there?

ytaskiran, 2021-04-30:
@LorenzMeier Actually I'd love to.  By the way, I think moving on with ROS2 instead of mavros would be more convenient from now on. What do you think?