PX4/PX4-Avoidance #133: [FEATURE] dis-/enable offboard control

TobiMiller, 2018-11-02:
Hey,

would be nice to have some handshake to control the drone with avoidance for a specific manner and to get back the offboard control when i need it. 

i will do it with a multiplexer node for the meantime.

best regards
tobi

mrivi, 2018-11-05:
Hi @TobiMiller, could you please explain a bit more what would you like to have? what do you mean by 

> to control the drone with avoidance for a specific manner

?

TobiMiller, 2018-11-09:
@mrivi 
i want to control the drone's velocity in a specific area where i know that there is no obstacle. but i want to turn on the the avoidance when i need it. 

for example: 
you have a big field with no trees. from there i want to fly into a forest (i know GPS accuracy and so on). i want to turn on the avoidance when i reach the beginning of the forest.

mrivi, 2018-11-09:
@TobiMiller A few more questions to understand your use case. 
Which algorithm are you using? local planner or global planner? 
Are you using the Offboard mode to control the vehicle in the obstacle free area? You mentioned a forest so I am assuming that you don't always have a global position available, is that correct?

TobiMiller, 2018-11-12:
@mrivi 
thanks for your answer. this was just an example. i have two areas. one where i want to control the velocity in offboard mode because i know that there are no obstacles (so there is GPS). One area where obstacles are present and where i want, that the global planner is doing his job.

mrivi, 2018-11-15:
@TobiMiller No, there is no support for what you're describing. It seems very specific to your use case. We assume no pre-existing knowledge about the environment.

nicovanduijn, 2018-12-04:
@TobiMiller I guess some of the confusion stems from the fact that the avoidance can use either offboard control or the mission interface to communicate with PX4. It sounds like @mrivi thought you simply wanted to change the mode of communication between PX4 and avoidance dynamically, which makes little sense in general.

Do I understand correctly that you wish to connect two companion computers to PX4, one running the avoidance and one to steer it (blindly) in offboard mode? That sounds dangerous...

If you are looking for a simple on/off switch for the avoidance, why not use the Dronecode SDK to modify the `MPC_OBS_AVOID` parameter?