PX4/PX4-Avoidance #161: Can't start the gazebo launch

LZMHIT, 2018-12-12:
Dear author,
       Thank you for your sharing firstly, however, When I start the launch "roslaunch global_planner global_planner_sitl_mavros.launch ", I got an error, it tells me that failed like the following picture.
[image]
Do you know how to deal with it.
And what is more, Can I use the gazebo groundtruth to replace gps so that I can simulator the indoor situation?
I am looking for your answer, Thank you very much!

nicovanduijn, 2018-12-12:
You should simulate the global planner with `roslaunch global_planner global_planner_stereo.launch`, as the `global_planner_sitl_mavros.launch` is an intermediate file included by the other launch files. We have a bit of a mess with launch files, I know...

mrivi, 2018-12-12:
@nicovanduijn I think you're getting confused with the local_planner. `roslaunch global_planner global_planner_stereo.launch` isn't an intermediate launch file.

@LZMHIT  the error `[gazebo-7] process has died` sometimes happens and you need to restart the simulation.
The error above it isn't coming from avoidance or PX4.

nicovanduijn, 2018-12-13:
@mrivi You're right, I was confusing it with `mavros_sitl.launch`. We really need to clean this up somehow

henkf, 2019-01-31:
You could try starting gazebo outside ROS by calling `gazebo` from the command line. If that fails, your gazebo is broken, make sure you are using the version supplied by your ROS distribution.

Next, you could try starting gazebo in a blank ROS session by starting `roscore`,  `rosrun gazebo_ros gzserver` and `rosrun gazebo_ros gzclient` in different terminals. If that fails, it is likely that your ROS environment and gazebo are incompatible.

It also looks like you are executing ROS as root, which should not be necessary and may hint at some confusion in your environment configuration.

closed 2019-02-18 by mrivi