PX4/PX4-Avoidance #162: runOdroid script global_planner

acpl00, 2018-12-12:
Hi people.

I am trying to run the runOdroid script from my host laptop to a Jetson TX2. 

In the beginning of the scrip there is  . ~/.bas_ros  and when the scrip reach the lines to  run the commands in separate tabs e.g. gnome-terminal --tab -e "ssh -t nvidia@10.42.0.1 'sleep 0;  . ~/.bash_ros; roscore; bash -i'" \

I get the error in the terminal bash: roscore: command not found.

My best guess is that it has to do with the line ". ~/.bas_ros"

Could some one help me :) thank you :)

mrivi, 2018-12-12:
Hi @acpl00 , you shoudn't run the script `runOdroid` if you are running the avoidance on TX2. Could you please give me some details about your setup such that I can point you towards the right scripts?
Are you using the local_planner or the global_planner? Which camera are you using?

acpl00, 2018-12-12:
I have tbs-discovery with a companion computer on board (Jetson TX2) and a stereo system ZED. Then I have my laptop which I want to control the drone over ssh. I run the runOdroid from my laptop. 

I practically follow the tutorial: 

https://github.com/NVIDIA-AI-IOT/redtail/wiki/Skypad-TBS-Discovery-Setup 

I would like to use a global planner since I read that is good enough for global navigation etc or which one you recommend :)

Thank you :)

mrivi, 2018-12-12:
@acpl00 I would recommend the local_planner. 
You can read about the HW setup here https://github.com/PX4/avoidance#run-on-hardware .
This [launch file](https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_A700_1cam.launch) can give you an example of all the components you need. However it's written for a Intel Realsense camera so you need to change [this part](https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_A700_1cam.launch#L42-L55)

mrivi, 2018-12-12:
@acpl00 the local planner has more flight hours. Furthermore, the gloabal planner builds a map of the environment so you need accurate position for it to work well. 

There isn't any ready to run launch file for the global planner. You can look at `local_planner_A7001cam.launch` to understand all the parts you need and make the necessary changes:
- local_planner -> global_planner
- realsense -> zed camera

acpl00, 2018-12-12:
Hi Martina,

Ok thank you, I will read and try the local planner. I will let you know how it goes :)

acpl00, 2018-12-13:
Hi Martina,

Good morning.

1.-The idea is to combine QGC-station with the global path planning, I mean, If we have the satellite view and the position of the drone, one can click in a certain distance on the map from the drone. Then, a square boundary will be created around the drone on the satellite view's map. Within that square area a  3D Octomap map is created, then in the empty space of that 3D map a kind of Voronoi map can be created, or in other words, there is a path in the middle of the obstacles so the drone is able to navigate. If the user apply roll and/or pitch the drone will move and once the user stops applying or moving the sticks of the RC transmitter, the drone will return to the main path.  While the drone is navigating in the empty space, the drone will fuse the 3D octomap map from the stereo vision with the one created from the satellite view.

2.-Do you think the global planning code can be used for this task? 

3.-How about the Odroidrun script? I read it was tested with the global planner, Am I right? Can I use that script to begin.

Thank you and looking forward to your replay :)

acpl00, 2018-12-13:
Ciao Martina,

Thank you so much for the info, it is quite helpful to start with.

Just two things from my previous mail.

1.-I want to show a 2D/3D occupied and empty space in Rviz from the Qgroundcontrol-station fly mission mode. So I do have an initial map. Do you think that is possible.

2.-In the RunOdroid script, there is a line  "gnome-terminal --tab -e "ssh -t odroid@odroid 'sleep 0;  . ~/.bash_ros; roscore; bash -i'" \"   that opens a terminal  an runs roscore, however I do not know what contains the file ~/.bash_ros; 

I know that script is set up for another copter, however I would like to know what it contains :) 

Cheers and regards,

mrivi, 2018-12-13:
1 - no, rviz will show you the environment that the vehicle has seen while flying and running the global planner.
2 - `bash_ros` is a script to set up the paths in the catkin workspace such as ROS_PACKAGE_PATH

acpl00, 2018-12-13:
Ciao Martina,

Would you mind sending me the content of the script :)

Thank you :)

mrivi, 2018-12-13:
you don't need it. If you follow the instruction on how to build the catkin workspace you will have everything working.

acpl00, 2018-12-13:
Hi Martina,

I have the catkin working space. however, I would like to take a look to the script, is that possible :)

Thank you :)

acpl00, 2018-12-13:
OK, thank you, I will take a look at what  you suggested :)

by the way, do you have experience with QGC-Station and fly-mission-plan? in case you have, when i set up all the points and I load the mission, I get the error 'Unable to start mission ('vehicle not ready')'.

The qgc has a gps locked, but I can see the GPS position is quite unstable. I guess that could be the reason.

cheers and thank you :)

mrivi, 2018-12-13:
It could be. Do you have any pre-flight checks warnings? Have you try flying first  in manual and then in position control?

TSC21, 2018-12-14:
@acpl00 please move your questions to the forum or to the Slack channels. The latest issue comments are not related with this topic anymore. Closing

closed 2018-12-14 by TSC21

acpl00, 2018-12-14:
Hi Nuno

Ok, I did not about that :) and how about if I have more questions about the related topic?

cheers

TSC21, 2018-12-14:
Reopen the issue and ask them. Though I think that @mrivi already gave you the required answers in https://github.com/PX4/avoidance/issues/162#issuecomment-446918450 and https://github.com/PX4/avoidance/issues/162#issuecomment-446931635

acpl00, 2018-12-15:
Hi Nuno,

Thank you for your replay.

I still have few questions about the global planning and I will open a new issue.

I mention them here 😊

1.- I guess that for mapping a SLAM is used, right? Just wondering what type of SLAM is used or a reference to the article would be great 😊

2.- What kind of path planning method is used?

3.- And how about the control control strategy, is a PID control strategy?

Thank you and wishing all the good.

Alfredo Ch??vez Plascencia.
Faculty of Information Technology (FIT),
Brno University of Technology
Czech Republic
00420603825858