PX4/PX4-Avoidance #166: global_planer

acpl00, 2018-12-15:
Dear people

I  have few questions about the global_planner

I mention them here 😊

1.- I guess that for mapping a SLAM is used, right? Just wondering what type of SLAM is used or a reference to the article would be great 😊 

2.- What kind of path planning method is used?

3.- And how about the control control strategy, is a PID control strategy?

Thank you and wishing all the good.

nicovanduijn, 2018-12-17:
1) Technically, no. SLAM is Simultaneous Localization And Mapping, the global planner does not do any localization, it relies on state estimate from the flight controller. It does build a map however, which is stored using an Octomap.
2) It's an A* search using a cost function that considers path length, risk of the path and smoothness.
3) Control is handled by PX4, the avoidance has nothing to do with it

@mrivi I have Villi's thesis on my laptop, can it be public? Might help the interested reader to put a link to it on the Readme. Same for @baumanta thesis.

acpl00, 2018-12-17:
Thank you for the answer,

I have tried the global planing gazebo-rviz. I have notice that it is difficult to choose a point destination when there is no initial map.

I have been thinking about  using rtabmap_ros for outdoor  map making. And,  during the global_planing  rtbmap  can  be used to fuse the estimated position with  the state estimate of the flight controller.

What do you think?

acpl00, 2019-01-26:
Hi 

Based what you have written I have few questions :) 
1.- I was thinking in case I  use rtabmap in the global map, how difficult would be to integrate GPS readings?

2.-What I understand is that the PX4 computes an EKF which is a kind of SLAM, right?  and this estimation is based on GPS, right?

3.-How about to integrate the estimation of the PX4 to the rtabmap to build up a map? do you think would be a good idea?

4.-The ZED camera contains already a odometry, and I am using that to build a 3D map, how about fusing that odometry with the estimation of PX4 and feed the rtabmap to build the map and later on fuse that map when using global planner?

cheers and looking forward for your replay?

jkflying, 2019-01-28:
Hello

PX4 doesn't do SLAM, and neither do the Global Planner. If you have a good source of position that can be used (eg. GPS) you can use the Global Planner to build a map, but it isn't very high quality since it is only designed to be accurate enough to do path planning. 

In terms of integrating these into PX4, a better place to ask would be at http://discuss.px4.io/, where people from other parts of PX4 will see your question and be able to answer.

closed 2019-01-28 by jkflying