PX4/PX4-Avoidance #177: [Err] [RTShaderSystem.cc:489] Unable to find shader lib. Shader generating will fail

Sunshinehualong, 2019-01-15:
## when I run  `make px4_sitl_default gazebo` 

 
[image]

mrivi, 2019-01-16:
`export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-7.0` . You need to change the gazebo version to match what you have installed

Sunshinehualong, 2019-01-17:
Thank you mrivi
It's really solved the problem.

closed 2019-01-17 by Sunshinehualong

ibrahimqazi, 2019-03-04:
@Sunshinehualong I have exactly the same error. Can you please guide me how to solve it? I have Gazebo multi-robot simulator, version 7.14.0

ibrahimqazi, 2019-03-04:
@mrivi can you please guide I have same error and my gazebo version is also 7.14.0.

nicovanduijn, 2019-03-04:
If you have exactly the same error, I would suggest exactly the same solution...?

ibrahimqazi, 2019-03-05:
@nicovanduijn he said to change the gazebo version, so change means upgrade or downgrade? By the way I am not PX4 package rather using another package but the problem is same as given above.

nicovanduijn, 2019-03-05:
It means you should set your `GAZEBO_RESOURCE_PATH` correctly. Depending on what version of gazebo  you are running.
If you are running gazebo 7.0, run `export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-7.0`.