PX4/PX4-Avoidance #179: Fast take-off in mission mode

nicovanduijn, 2019-01-16:
During our tests we found that the mission take-off is violently fast when the avoidance is running. This may have gone unnoticed for a long time because we usually take off in position mode before starting the mission.

Sadly, we didn't have logging from boot enabled, so it is not completely captured in the logs below. But we can see that without the avoidance, the beginning of the log still shows the end of a smooth ramp in the z-setpoint, while the log with the enabled OA starts at a 3m altitude (to which it got presumably through a step)

No avoidance:
[image]

Avoidance enabled:
[image]

mrivi, 2019-01-16:
@nicovanduijn  I don't know if you are already aware that when OA is running is completely overwrites the auto takeoff coming from the Firmware hence the completely different behaviour
OA auto take off https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/waypoint_generator.cpp#L202
Firmware auto takeoff https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/AutoMapper2/FlightTaskAutoMapper2.cpp#L135

nicovanduijn, 2019-01-16:
yes, I'm aware of that. What I'm saying is that the OA's takeoff should be much smoother than it currently is

closed 2019-03-04 by baumanta