PX4/PX4-Avoidance #214: Global_planner pixhawk estimated pose

acpl00, 2019-02-12:
Dear people.

I am on the way thinking about implementing the global planner in real.

I can not figure out where in the mavros launch file is specified the transformation between  NED->ENU conversion.

In other words, I would like  to set up in my own mavros launch file  the coordinates NED-Pixhawk  aligned or transformed with the coordinates ENU-ROS.

Any help I appreciate it.

Thank you.

mrivi, 2019-02-12:
@acpl00 the conversion is done by Mavros. This is how you should launch mavros
https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_A700_1cam.launch#L20-L30 
You need to change the `fcu_url` to match your setup. 

The transformation you need to specify directly in the launch file is the one between local_origin and the camera frame. See example here https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_A700_1cam.launch#L16

acpl00, 2019-02-12:
Hi Mrivi

Thank you for the replay.

In the line <arg name="config_yaml" value="$(find local_planner)/resource/px4_config.yaml" />

Mavros comes with its own px4_config.yaml, right?  and the one you are using in the local planner is setup to your own drone, right?

In my case I shall use the one from mavros, right?

mrivi, 2019-02-12:
@acpl00  no, it's vehicle independent. You should use the one in `global_planner/resources`.

acpl00, 2019-02-12:
OK, thank you,

and how about this lines in the mavros launch file:

            <param name="/mavros/local_position/tf/send"       type="string" value="true"/>
             <param name="/mavros/global_position/tf/send"      type="string" value="true"/>

are they necesary to bring up the transformations?

acpl00, 2019-02-12:
Hi again,

in my case, would be:
 <node pkg="tf" type="static_transform_publisher" name="tf_front_camera"
          args="0.30 0.32 -0.11 0 0.052 0 fcu zed_center 10"/>

Just need to see the rotation between fcu and zed_center to set up correctly the transformation, right?

acpl00, 2019-02-12:
I may also need a transformation between either /map->/local_origin, right?

It could be something like this, right? 

node pkg="tf" type="static_transform_publisher" name="map_to_local" args="0.0 0.0 0.0 0.0 0.0 0.0 /map /local_origin 100" /

acpl00, 2019-02-12:
Hi Again,

I have set up the mavros as you suggested without any static transformation to see how they look like.

well I get two trees

1.- /map -> /odom -> /based_link -> /zed_center -> (a.- /zed_left_camera and b.- /zed_right_camera)

2.- /local_origin -> /fcu -> /fcu_frd
     /local_origin -> /local_origin_ned

The only frame which is rotated is the /local_origin_ned (NED) with respect to /map (ENU)

So, I am in doubt how to set up the static transformation that I can relate the two trees.

Any help I appreciate it.

Thank you.

acpl00, 2019-02-14:
Dear people,

I have some doubts about about the  global planer px4_config.yaml file.

1.- The frame_id of the imu is the fcu frame. Is the fcu the main frame of the pixhawk? can I change the imu's fcu frame  with the base_link?

2.- What is the main frame? fcu or local origin?

3.-  In which frame  is given the estimated pose of the pixhawk?

4.-Is this pose the global pose or the local pose?

Thank you very much

mrivi, 2019-02-18:
@acpl00 Mavros publishes the vehicle position in the `local_origin` frame. The planner reference frame is `world` which is the same as the `local_origin` frame ([here](https://github.com/PX4/avoidance/blob/master/global_planner/launch/global_planner_sitl_mavros.launch#L18-L19) you can see the transformation between the two frames). The transformation between the two frames is handled in the callbacks for the relevant data.

acpl00, 2019-02-20:
Thank you,

and just wondering what is the role of fcu and fcu_frd in the px4_config.yaml

acpl00, 2019-02-20:
Hi again, do you know how can I get the IMU frame so I can see it in rviz?

Thank you

mrivi, 2019-02-20:
`fcu` `fcu_frd` transformation from mavros https://github.com/mavlink/mavros/blob/master/mavros/src/lib/uas_data.cpp#L56

You can visualize the `fcu` frame when you choose the axes visualization in rviz

acpl00, 2019-02-20:
Thank you, 

So the frame_id  of the imu   in the original px4_config.yaml of  mavros is base_link but in the px4_config.yaml of the global planner is fcu. 
  
This make me think that base_link can be used instead of fcu, right?

acpl00, 2019-02-22:
Hi again :)

I have noticed that in RVIZ, the  fcu goes in total different direction than base_link or in other words the drone direction.   Do you know why is that?

Thank you

acpl00, 2019-03-13:
Hi Mrivi 

You said that Mavros publishes the vehicle position in the local_origin frame.
But the question here is which topic of  mavros has the drone position:

I get
 /mavros/global_position/local
/mavros/local_position/odom

My best guess is that    "/mavros/global_position/local"  has the position from the GPS that is measure in the local_position_ned frame. I am right?

Or how is it?

Thank you :)

mrivi, 2019-03-14:
@acpl00 the global planner doesn't subscribe to neither of those topics but `/mavros/local_position/pose` https://github.com/PX4/avoidance/blob/08dcfd01700f8570fd4f7da99b1fbb150c016f40/global_planner/src/nodes/path_handler_node.cpp#L24 .

If you are interested in the differences between the topics please have a look at
- https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp
- https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/local_position.cpp

acpl00, 2019-03-14:
Hi mrivi

Thank you for your replay.

As far as I can see` /mavros/local_position/pose `gives me the `xyz` of the position where as the `/mavros/local_position/odom` gives me the same position as `/mavros/local_position/pose` plus covariance.

I have made a measurement test, where I plot in `RVIZ`  the `/mavros/local_position/pose` and a `odom` topic based on `rtabmap.`  

[image]

The `rtabmap` which is the colored line gives me the trajectory of the drone whereas the white path is the `/mavros/local_position/pose`. I am wondering why is the shift and rotation.

I want to use` /mavros/local_position/pose` topic but how can use it   if it  is not correct, e,g, it is shifted and rotated.  Do you have a clue how can this problem be solved?

Thank you.

acpl00, 2019-03-17:
Thank you, 

So the frame_id  of the imu   in the original px4_config.yaml of  mavros is base_link but in the px4_config.yaml of the global planner is fcu.

closed 2019-03-17 by acpl00