PX4/PX4-Avoidance #217: make px4_sitl_default gazebo fails to build

magiccreator69, 2019-02-12:
Hello,
I'm a beginner in px4, ROS, and Gazebo. My target is to deploy this program on jetson tx2. But before that, I want to try on my desktop. Followed the steps mentioned in this repository and installed all the dependencies. After I do "make px4_sitl_default gazebo", it opens a new Gazebo window and freezes making the program at "EKF commencing GPS fusion". I got no idea where should I start from as I can't run the program unless I've built it successfully. Am I supposed to create obstacles in the Gazebo simulator myself and provide the file path to the program? Thanks in advance.

[image]

[image]

dagar, 2019-02-12:
You should be seeing a quadcopter (3dr iris) in the Gazebo window.

dagar, 2019-02-12:
@TSC21 any ideas?

magiccreator69, 2019-02-13:
Yes, I can see the quadcopter which is comparatively small in size that's why can't be seen in the picture I've attached here. What do I need to do now?

mrivi, 2019-02-13:
@magiccreator69 if you have the ROS already installed you need to follow the instructions on how to run the local or global planner (https://github.com/PX4/avoidance#run-the-avoidance-gazebo-simulation)

If you don't have ROS installed please follow [these instructions](https://github.com/PX4/avoidance#installation-for-ubuntu-1604-and-ros-kinetic)

julianoes, 2019-04-24:
This looks stale, please re-open if it's still an issue.

closed 2019-04-24 by julianoes

mta32, 2020-07-19:
Was the issue resolved ?? Where you able to simulate obstacle avoidance after adding an obstacle