PX4/PX4-Avoidance #227: Global Planner: Empty Path

aydal, 2019-02-16:
Hi community,

I have recently began working on global planner and tried to push the maximum altitude achievable. I edited the value in configuration files (global_planner/cfg). I delved into the code and tried to tweak parameters. Now, while the max_altitude is 200m, following is happening.
1. Empty path for anything above 50m
2. Once I reach 50 m, then i publish new altitudes in increments. Still, it falters at 90m.

Any kind of help or guidance is much appreciated.

mrivi, 2019-02-18:
@aydal there shouldn't be any altitude limitation unless you have geofence enabled.

ramunter, 2019-02-18:
Could there be an issue with the `alt_prior_` found in `global_planner.h`? 

The `getAltPrior` function index's `alt_prior_` using `cell.zPos()`, so the code should crash if the zPos is higher than the length of `alt_prior_`

mrivi, 2019-02-20:
@ramunter your're correct. 
@aydal Does the planner crash or what's the behaviour you're seeing?

aydal, 2019-02-20:
I have attached the terminal output for global_planner_sitl_mavros.launch.
I altered global_planner_node to use move_base's (x,y,z) rather than just (x,y). Point given is (10,10,100)

[image]

mrivi, 2019-02-20:
Is there a particular reason why the vehicle should fly at 100m? 
I would suggest lowering the altitude such that the planner works. At the moment the dev team is focused on the local planner so if you have a hard requirement of 100m altitude on the gloabl planner a PR is more than welcomed.

aydal, 2019-02-20:
The planner works erratically at 50m. There is no exact reason behind testing the drone at 100m as such. I am just looking into the vulnerabilities of global planner which may compromise the robustness and wish to contribute to this open source community. 
My aim is to integrate avoidance with QGC so that we can have obstacle avoidance by default. So that's why I am tweaking the code and parameters to single out all problems. And if I have avoidance added as default planner it will be problematic if it just shuts down at high altitudes. 
Other things I am looking into our optimisation and a wide array of sensor integration for octomap server and QGC integration. Help will be appreciated on that aspect as well

mrivi, 2019-02-20:
@aydal have you looked into the `alt_prior_` as @ramunter suggested? You might need to extend that array to make the planner work at altitudes higher  than ~25m. Contributions are really appreciated!
Could you explain what do you mean with QGC integration? There is ongoing work on tightly integrating an avoidance system into PX4 so maybe there is already something in the works for your application.

aydal, 2019-02-20:
I will look into it. Thanks

closed 2019-07-04 by jkflying