PX4/PX4-Avoidance #234: can not changed OFFBOARD mode when running local_planner

moonlive, 2019-02-22:
I am now trying to run the local planner as a simulator.
roslaunch local_planner local_planner_depth-camera.launch
If you do this, rviz will be executed and the map will be displayed correctly in rviz.

If you run the "rosrun mavros mavsys mode -c OFFBOARD" command from another terminal window
"Request failed. Check mavros logs" message is displayed.
On the screen where roslaunch was run
"[ERROR] [1550808496.159953057, 32.364000000]: MODE: Unsupported FCU"
A message is output.
It does not work with QGC either.

If you check "rostopic echo -n2 / diagnostics", the following message appears.
header: 
  seq: 115
  stamp: 
    secs: 110
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 1
    name: "mavros: FCU connection"
    message: "not connected"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Received packets:"
        value: "0"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "0"
      - 
        key: "Tx sequence number:"
        value: "39"
      - 
        key: "Rx total bytes:"
        value: "0"
      - 
        key: "Tx total bytes:"
        value: "31497"
      - 
        key: "Rx speed:"
        value: "0.000000"
      - 
        key: "Tx speed:"
        value: "287.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 2
    name: "mavros: Heartbeat"
    message: "No events recorded."
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Vehicle type"
        value: "Generic micro air vehicle"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: ''
      - 
        key: "System status"
        value: "Uninit"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
header: 
  seq: 116
  stamp: 
    secs: 111
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 1
    name: "mavros: FCU connection"
    message: "not connected"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Received packets:"
        value: "0"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "0"
      - 
        key: "Tx sequence number:"
        value: "52"
      - 
        key: "Rx total bytes:"
        value: "0"
      - 
        key: "Tx total bytes:"
        value: "31800"
      - 
        key: "Rx speed:"
        value: "0.000000"
      - 
        key: "Tx speed:"
        value: "303.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 2
    name: "mavros: Heartbeat"
    message: "No events recorded."
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Vehicle type"
        value: "Generic micro air vehicle"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: ''
      - 
        key: "System status"
        value: "Uninit"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor helth"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

I think the problem here is that the "mavros: FCU connection" item appears as "not connected".
But I do not know how to solve this.
What can i do?

ps : "make px4_sitl_default gazebo" it work well.

mrivi, 2019-02-22:
@moonlive have you followed [these steps](https://github.com/PX4/avoidance#run-the-avoidance-gazebo-simulation) ?
Can you share your `ROS_PACKAGE_PATH`, `GAZEBO_MODEL_PATH`, and `GAZEBO_RESOURCE_PATH` environment variables?

moonlive, 2019-02-22:
@mrivi  Sure.

ROS_PACKAGE_PATH=/home/moonlive/catkin_ws/src/mavlink:/home/moonlive/catkin_ws/src/mavros/libmavconn:/home/moonlive/catkin_ws/src/mavros/mavros_msgs:/home/moonlive/catkin_ws/src/mavros/mavros:/home/moonlive/catkin_ws/src/avoidance/global_planner:/home/moonlive/catkin_ws/src/mavros/mavros_extras:/home/moonlive/catkin_ws/src/avoidance/local_planner:/home/moonlive/catkin_ws/src/modudculab_ros:/home/moonlive/catkin_ws/src/ros_tutorials-melodic-devel/ros_tutorials:/home/moonlive/catkin_ws/src/ros_tutorials-melodic-devel/roscpp_tutorials:/home/moonlive/catkin_ws/src/ros_tutorials-melodic-devel/rospy_tutorials:/home/moonlive/catkin_ws/src/rplidar_ros:/home/moonlive/catkin_ws/src/mavros/test_mavros:/home/moonlive/catkin_ws/src/ros_tutorials-melodic-devel/turtlesim:/opt/ros/kinetic/share:/home/moonlive/Firmware

GAZEBO_MODEL_PATH=:/home/moonlive/catkin_ws/src/avoidance/sim/models:/home/moonlive/Firmware/Tools/sitl_gazebo/models

GAZEBO_RESOURCE_PATH is not set.
Could it the reason of problem?

mrivi, 2019-02-22:
@moonlive  If the models are loaded correctly it's fine as it is. What about `GAZEBO_PLUGIN_PATH` and `LD_LIBRARY_PATH` ?

moonlive, 2019-02-22:
@mrivi  Yes it is. There was no error message when launch was run.

GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-8/plugins

LD_LIBRARY_PATH=/home/moonlive/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu

mrivi, 2019-02-22:
@moonlive run this `. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default` script. You should have the PX4 Firmware build folder in those paths

moonlive, 2019-02-23:
@mrivi Thank you for your interest and advice.
 I did that. but that was not solution of my problem.

I checked the output message through the [rostopic echo -n2 / diagnostics] command.
As a result I guess that it is not connected to px4_sitl_default. It seems can not receive heartbeat.
Could you show me the way what should i do to find clues?

moonlive, 2019-02-23:
@mrivi I did it!! Thank you so much.

I ran ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default. but it does not apply to environments value.
I don't know why.. 
Anyway I did set as same values setup_gazebo.bash of GAZEBO_PLUGIN_PATH, GAZEBO_MODEL_PATH and LD_LIBRARY_PATH .
And then, It is work.

So, i close this issue.

closed 2019-02-23 by moonlive