PX4/PX4-Avoidance #238: local planner on Jetson TX2 with PX4

magiccreator69, 2019-03-04:
Hello Everyone, I'm a total beginner in ROS, PX4, and gazebo. Actually, don't have much knowledge about all this stuff. I'm just learning as much as I can from here. I've followed all the instructions mentioned here on this repository. So far I've got to the hardware implementation part. Let me break down the steps I've tried and the subsequent error messages/problems faced.
Hardware Setup:
CPU: Intel Core i7-7700K@4.20GHz x 8
GPU TITAN X (Pascal)-12 GB
PX4 Flight Controller Unit (FCU) - Both official and customized
Companion Computer: Jetson TX2
Camera: Intel Realsense D435i
Installed Software:
ROS-Kinetic-Desktop-Full
Intel Realsense SDK: v2.18.0
ROS-Realsense Node: 2.1.4

I've run the global_planner node on my desktop setup couple of times. A couple of times I could move the drone by pointing at a coordinate from RVIZ. But now the map isn't loaded in RVIZ. 
[image]
[image]
[image]
[image]

I've also successfully installed all the dependencies and required software packages on Jetson TX2. can establish the connection between px4 and the companion computer which is Jetson TX2. But when I run the local planner following the methods mentioned here in this repo, the world isn't loaded in rviz and gazebo. What am I supposed to do now? Looking forward to your suggestions. Thanks in advance.

[image]
[image]
[image]

nicovanduijn, 2019-03-04:
I believe you're mixing up the SITL (Software-in-the-loop) with real hardware tests.
What exactly are you trying to accomplish? Your machine is plenty powerful for SITL, so let's get you a running simulation environment first. Can you run something like `roslaunch local_planner local_planner_depth-camera.launch` without errors? What if you create a mission in QGC, upload and run it?

magiccreator69, 2019-03-05:
Thanks for your prompt reply. Actually, I've tried the simulation on my desktop setup and run the real hardware tests on jetson tx2 with px4 along with intel realsense camera exactly as mentioned in this repository. I'm just a beginner in this field. I'm trying to run the avoidance program and understand how it works on UAV. right, forgot to mention something! as Intel doesn't provide sdk for arm architecture, I followed jetsonhacks git repository( https://github.com/jetsonhacks/installRealSense2ROSTX ) to install intel realsense sdk(v2.18.0) and realsense ros-wrapper(2.1.4). The camera seems to be working fine with jetson tx2. 
Yes, I can upload a mission in QGC but it somehow doesn't connect with other ros nodes.
Just now I ran `roslaunch local_planner local_planner_depth-camera.launch`  on my desktop setup and this is what I get:

`... logging to /home/sani/.ros/log/0209435a-3edd-11e9-bf35-2c4d5497ea54/roslaunch-sani-desktop-5435.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sani-desktop:40287/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /factor_close_to_goal_start_speed_limitation: 3.0f
 * /factor_close_to_goal_stop_speed_limitation: 4.0f
 * /goal_acceptance_radius_in: 0.5f
 * /goal_acceptance_radius_out: 1.5f
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_node/world_name: /home/sani/catkin...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
[... 178 lines cut ...]
[ INFO] [1551745476.692825381]: Plugin wind_estimation loaded
[ INFO] [1551745476.693344562]: Plugin wind_estimation initialized
[ INFO] [1551745476.693371099]: Autostarting mavlink via USB on PX4
[ INFO] [1551745476.693399306]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1551745476.693415192]: Built-in MAVLink package version: 2019.2.2
[ INFO] [1551745476.693433487]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1551745476.693450698]: MAVROS started. MY ID 1.240, TARGET ID 1.1
SpawnModel script started
[INFO] [1551745476.909913, 0.000000]: Loading model XML from Gazebo Model Database
[INFO] [1551745476.910078, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1551745477.286662547]: Stereo is NOT SUPPORTED
[ INFO] [1551745477.286713756]: OpenGl version: 4.5 (GLSL 4.5).
[ INFO] [1551745477.736228090, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1551745477.761936286, 0.048000000]: Physics dynamic reconfigure ready.
[INFO] [1551745477.816834, 0.100000]: Calling service /gazebo/spawn_sdf_model
[ INFO] [1551745478.168668638, 0.120000000]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1551745478.170389202, 0.120000000]: Camera Plugin (ns = )  <tf_prefix_>, set to ""
[INFO] [1551745478.398351, 0.120000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1551745478.402592292, 0.120000000]: Laser Plugin: The 'robotNamespace' param was empty
[ INFO] [1551745478.402636183, 0.120000000]: Starting Laser Plugin (ns = iris)
[ INFO] [1551745478.403193683, 0.120000000]: Laser Plugin (ns = iris)  <tf_prefix_>, set to "iris"
[spawn_model-6] process has finished cleanly
log file: /home/sani/.ros/log/0209435a-3edd-11e9-bf35-2c4d5497ea54/spawn_model-6*.log
`
[image]

baumanta, 2019-04-09:
Is this still an issue? Or can we close it?