PX4/PX4-Avoidance #243: erros when launch gazebo

FPSychotic, 2019-03-07:
when i execute the .launch of the installation example, it send me this two error, but I can see the houses, but the grid of rviz is empy there is not the black blocks that in the picture of the example.
it is running in 18.04 and melodic,mavros staff from debian pkg.ROS" is installed but melodic is sourced.
i detected if print the echo $GAZEBO_PLUGIN_PATH

send me a line empty

THANKS!

ERROR: cannot launch node of type [global_planner/global_planner_node]: can't locate node [global_planner_node] in package [global_planner]

ERROR: cannot launch node of type [global_planner/path_handler_node]: can't locate node [path_handler_node] in package [global_planner]

mrivi, 2019-03-07:
@FPSychotic try building the planner again and sourcing the ROS envirnment variables
```bash
catkin build global_planner
source devel/setup.bash
```

FPSychotic, 2019-03-07:
i detect my variables didnt be seted, I dont kmow why. I go that rviz show the blocks in he grid like in the picture.
anyway I rebuildt global_planner and simulation and I got the sam errors.

the only thing I did different is use catkin_make as my catkin_ws is not empty and was build with catkin_make, and Im nor sure if I can use catkin build. Anyway no compiling errors

mrivi, 2019-03-07:
@FPSychotic  can you share the output of  `echo $ROS_PACKAGE_PATH`?

FPSychotic, 2019-03-08:
thanks for the help, it is working after compile it with catkin build as the guide explain, I did with catkin make. 
anyway is strange as after worked yesterday I needed compile it again today. it send me other error, but it looks work

closed 2019-03-08 by FPSychotic

bys1123, 2019-04-17:
Thanks for this information!