PX4/PX4-Avoidance #247: global planner

PrasaNth2897, 2019-03-08:
when i tried to run the global planner comment
roslaunch global_planner global_planner_sitl_mavros.launch
it displaces an error like this:
[global_planner_sitl_mavros.launch] is neither a launch file in package [global_planner] nor is [global_planner] a launch file name
The traceback for the exception was written to the log file
[image]
can you please correct  my error or what is the next step should i do

mrivi, 2019-03-09:
try 
```bash
cd ~/catkin_ws/
source devel/setup.bash
roslaunch global_planner global_planner_sitl_mavros.launch

PrasaNth2897, 2019-03-10:
gazebo has not opened 
[image]
[image]
[image]

please rectify my error

prasanth@prasanth-VirtualBox:~/catkin_ws$ roslaunch global_planner global_planner_sitl_mavros.launch
... logging to /home/prasanth/.ros/log/3e3fbd92-42eb-11e9-814d-080027a4f152/roslaunch-prasanth-VirtualBox-3123.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: px4
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/prasanth/catkin_ws/src/avoidance/global_planner
ROS path [2]=/home/prasanth/catkin_ws/src/avoidance/local_planner
ROS path [3]=/opt/ros/kinetic/share

mrivi, 2019-03-10:
@PrasaNth2897 you haven't set the environment variables correctly. Please make sure that you have built the PX4 firmware (`make px4_sitl_default gazebo`). Then run 

```bash
. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models

PrasaNth2897, 2019-03-12:
@mrivi 
i tired to launch gazebo
roslaunch global_planner global_planner_sitl_mavros.launch

but in it shows global status error and in gazebo it shows only the drone
[image]
[image]

can you plz rectify this

PrasaNth2897, 2019-03-12:
and also the log shows some error while the ros launch comment is executed
[image]
[image]

closed 2019-07-01 by mrivi