PX4/PX4-Avoidance #248: Spawn service failed. Exiting.

FPSychotic, 2019-03-08:
this is the log of that error. the simulation look works, I dont know to what reference or aim this error. 18.04 melodic up board x86 intel graphics 

[rosout][INFO] 2019-03-08 21:16:00,966: Loading model XML from Gazebo Model Database
[rosout][INFO] 2019-03-08 21:16:00,974: Waiting for service /gazebo/spawn_sdf_model
[rospy.internal][INFO] 2019-03-08 21:16:01,189: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2019-03-08 21:16:03,991: topic[/clock] adding connection to [http://ihexa-UP-CHT01:44723/], count 0
[rosout][INFO] 2019-03-08 21:16:04,029: Calling service /gazebo/spawn_sdf_model
[rosout][INFO] 2019-03-08 21:16:04,954: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris
[rosout][ERROR] 2019-03-08 21:16:04,958: Spawn service failed. Exiting.
[rospy.core][INFO] 2019-03-08 21:16:04,965: signal_shutdown [atexit]
[rospy.internal][INFO] 2019-03-08 21:16:04,981: topic[/rosout] removing connection to /rosout
[rospy.internal][WARNING] 2019-03-08 21:16:04,988: Unknown error initiating TCP/IP socket to ihexa-UP-CHT01:47101 (http://ihexa-UP-CHT01:44723/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 559, in connect
    self.local_endpoint = self.socket.getsockname()
AttributeError: 'NoneType' object has no attribute 'getsockname'

[rospy.internal][INFO] 2019-03-08 21:16:04,989: topic[/clock] removing connection to http://ihexa-UP-CHT01:44723/
[rospy.impl.masterslave][INFO] 2019-03-08 21:16:04,990: atexit

FPSychotic, 2019-03-08:
i think that globall planner is looking by iris_depth_camera in the ros7melodic/gazebo...but I found that file in catkin_ws
maybe that can help to track the problem

mrivi, 2019-03-09:
@FPSychotic  make sure that the models folder is in your GAZEBO_MODEL_PATH
`export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models`

FPSychotic, 2019-03-09:
yes, the variable is right, it is in my bash.rc with the others. the thing is that all load slow, but finally I can see a depth camera in rqt_reconfigure (so, did lit load properly I guess?) and the drone avoid obstacles and rviz show the occupancy map. I can run the local planner example with the stereo camera too, maybe the service fails due to the slow load and is called again, and load properly?
other thing is in melodic there is not yaml pkg to can follow the guide, I didnt installed that pkg, could be that?

mrivi, 2019-03-09:
@FPSychotic what exactly isn't working?

FPSychotic, 2019-03-09:
Im not sure, I recive the error of the 1st message, I dont know what mean
that error so I cant to know if something dont work.
That error refer the depth camera?

closed 2019-03-28 by FPSychotic

LeopoldoCespedes, 2023-04-14:
>va a thread de gazebo a solucionar problema gazebo
> random1 : "ni puta idea de como solucionarlo"
>cierran la thread
mfw
[image]