PX4/PX4-Avoidance #251: Failed to load nodelet [/camera/points_xyz_sw_registered]

moonlive, 2019-03-14:
Hello every one.

I try avoidance function as TX2 and PX4.

Companion PC : Jetson TX2
Camera : Intel Realsense D435
OS :Ubuntu 16.04
ROS : ros-kinetic-ros-base
mavros version : 0.29.0
cmake version : 3.13.4
Realsense SDK 2.0

I ran roslaunch as below.
roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyTHS2:921600
FC is connected by ttyTHS2 serial port.

message is below.

-----------------------------------------------------------------------------------------------------------------
started roslaunch server http://tegra-ubuntu:43328/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 5
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_imu: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 270
[... 311 lines cut ...]
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
         at line 126 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
---------------------------------------------------------------------

What can i do for solving nodelet errors?
Please show me the solving way or clue.
Thanks.

baumanta, 2019-03-15:
I would do the following to debug:
- try starting only the camera `roslaunch realsense2_camera rs_rgbd.launch` do you see the topics? Are they published at the right rate? If that fails you might have a problem with your camera and you should contact Intel
-try starting the Realsense with the postprocessing node for the depth cloud  `roslaunch local_planner rs_depthcloud.launch` Is the pointcloud topic published? If not there might be a problem with the postprocessing node, it is part of the [depth_img_proc package](http://wiki.ros.org/depth_image_proc)
-once both cammands work you should be able to start the avoidance by `roslaunch local_planner avoidance.launch`

moonlive, 2019-03-16:
@baumanta 
Thank you for your reply.

I did `roslaunch relsense2_camera rs_rgbd.launch` and got the message like below.

--------------------------------------------------------------------------
> nvidia@tegra-ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_rgbd.launch 
... logging to /home/nvidia/.ros/log/aecda338-4798-11e9-962a-00044bc5cb2d/roslaunch-tegra-ubuntu-18699.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 655, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/nvidia/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
ROS path [3]=/opt/ros/kinetic/share
ROS path [4]=/home/nvidia/catkin_ws/src/disparity_to_point_cloud
ROS path [5]=/home/nvidia/catkin_ws/src/mavlink
ROS path [6]=/home/nvidia/catkin_ws/src/mavros
ROS path [7]=/home/nvidia/catkin_ws/src/realsense-2.2.1
nvidia@tegra-ubuntu:~/catkin_ws$ 
----------------------------------------------------------
I think `ResourceNotFound: rgbd_launch` is the reasone of this problem. right?
Almost working python files are /opt/ros/kinetic/lib/python2.7/dist-packages/ 
but last one is /usr/lib/python2.7/dist-packages/rospkg/.
Is it problem?

What can i do for fix it?
Thank you.

Jaeyoung-Lim, 2019-03-16:
@moonlive  You seem to be missing `rgbd_launch` Have you tried installing it?
```
apt install ros-kinetic-rgbd-launch
```

moonlive, 2019-03-17:
Thank you @Jaeyoung-Lim.
It dosen't install.
I make install and check the error message is gone.
But I don't know whether it work well or not.
I didn't check  '[OA] Planning started, using 1 camera' message.
Below  "UVC metadata payloads not available" message is the cause ?
I attach log message. 
Please check.
------------------------------------------------------------------------------------------------------
[ INFO] [1552803133.985967222]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1552803133.991114057]: insert Depth to Stereo Module
[ INFO] [1552803133.991304615]: insert Color to RGB Camera
[ INFO] [1552803133.991388486]: insert Infrared to Stereo Module
[ INFO] [1552803133.991440134]: insert Infrared to Stereo Module
 17/03 15:12:14,564 WARNING [547734155664] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1552803134.594367584]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
[ INFO] [1552803136.912664958]: Add Filter: pointcloud
[ INFO] [1552803136.924087481]: num_filters: 1
[ INFO] [1552803136.924203000]: SELECTED BASE:Depth, 0
[ INFO] [1552803136.929215244]: RealSense Node Is Up!
[ INFO] [1552803136.929302731]: Setting Dynamic reconfig parameters.
[ INFO] [1552803136.962052810]: Done Setting Dynamic reconfig parameters.
 17/03 15:12:17,018 WARNING [547686338960] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
----------------------------------------------------------------------------------------------------------

Jaeyoung-Lim, 2019-03-17:
@moonlive What do you mean it doesn't install? Does `ros-kinetic-rgbd-launch` not get installed?

mrivi, 2019-03-17:
`rs_rgbd.launch` is a launch file in the `realsense2_camera` package. (https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/rs_rgbd.launch) 

@moonlive please make sure that you have built that package (`catkin build realsense2_camera`) and that you have sourced your catkin environment (`source ~/catkin_ws/devel/setup.bash`)

moonlive, 2019-03-18:
Hello @Jaeyoung-Lim 
I mean.. I didn't install `ros-kinetic-rgbd-launch`. So i install it as your advice and dispared the error message.
But i don't know it works well.
한국분이신것 같아 다시 간단하게, 설치 하지 않았었구요 설치후에 그 에러 메시지가 사라진것 확인 했습니다.
하지만 여전히 제대로 동작 하는것 같지는 않네요.
감사합니다.

moonlive, 2019-03-18:
Hello @mrivi 
Situation changed after I installed `ros-kinetic-rgbd-launch`.
When I run avoidance.launch and rs_rgbd.launch, I get the same warning message.
The message is ` Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`.

moonlive, 2019-03-18:
Hello everyone.

I found `https://github.com/freemanlo/librealsense`.
What is different with `https://github.com/intel-ros/librealsense-release`?
Could I use `https://github.com/freemanlo/librealsense` for jetson?

Thank you.

mrivi, 2019-03-18:
@moonlive It's a fork of the official intel repository 

 Could you share the entire output of `avoidance.launch` or `rs_rgbd.launch` ?

moonlive, 2019-03-19:
@mrivi 
I'm sorry.  I tring reinstall evironmets of my TX2 now.
I will upload log after finish work it.

moonlive, 2019-03-19:
Hi @mrivi

I tried install environments again.
output of 'roslaunch realsense2_camera rs_rgbd.launch' 

----------------------------------------------------------------------
... logging to /home/nvidia/.ros/log/e54b3d0c-4a50-11e9-b664-00044bc5cb2d/roslaunch-tegra-ubuntu-28459.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:46611/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [28469]
ROS_MASTER_URI=http://localhost:11311

[... 26 lines cut ...]
[ INFO] [1553004693.725822957]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1553004693.725892972]: setupPublishers...
[ INFO] [1553004693.732772102]: Expected frequency for depth = 30.00000
[ INFO] [1553004693.740050107]: Expected frequency for infra1 = 30.00000
[ INFO] [1553004693.744256381]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1553004693.748955415]: Expected frequency for infra2 = 30.00000
[ INFO] [1553004693.753140633]: Expected frequency for aligned_depth_to_infra2 = 30.00000
[ INFO] [1553004693.757581176]: Expected frequency for color = 30.00000
[ INFO] [1553004693.763128646]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1553004693.772854262]: setupStreams...
 19/03 14:11:33,787 WARNING [548123746304] (sensor.cpp:329) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1553004693.794933200]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553004693.795620358]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553004693.796245309]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 19/03 14:11:33,801 WARNING [548123746304] (backend-v4l2.cpp:1160) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 19/03 14:11:33,803 WARNING [548123746304] (sensor.cpp:329) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1553004693.817814590]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553004694.047610331]: publishStaticTransforms...
[ INFO] [1553004694.047896503]: RealSense Node Is Up!
 19/03 14:11:34,072 WARNING [547601240464] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1553004694.082585751]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 19/03 14:11:34,220 WARNING [547590377872] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 19/03 14:11:34,561 ERROR [548123746304] (types.h:179) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1553004694.561898543]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
---------------------------------------------------------------------------

mrivi, 2019-03-19:
@moonlive What kernel version do you have?
Have a look at what's suggested here https://github.com/IntelRealSense/librealsense/issues/2029

moonlive, 2019-03-20:
@mrivi 
kernel version is 4.4.38-tegra

I did reboot and retry the launch.
`Frame metadata isn't available! ` message was gone.
Now is  last only one warning message it `Unregistered Media formats : [ UYVY ]; Supported: [ ]`

mrivi, 2019-03-20:
@moonlive From your output that seems to be a warning. Can you check if you get a point cloud or depth image in Rviz?
If you don't, try relaunching the launch file several times and also powercycling the realsense. My experience is that the camera node fails more times than it works.

moonlive, 2019-03-20:
@mrivi 
Yes I can see images as `rs-pointcloud`
I did try many times turn off-on camera but can not go further.

Could it cause of problem?
Enable PointCloud: Off
Align Depth: Off

local_planner node is not work.
My ROS is ros-kinetic-desktop. Could I install ros-kinetic-desktop-full ?

----------------------------------------------------
[ INFO] [1553070463.397602996]: ROS Node Namespace: camera
[ INFO] [1553070463.397718650]: Device Name: Intel RealSense D435
[ INFO] [1553070463.397791585]: Device Serial No: 832112073135
[ INFO] [1553070463.397866279]: Device FW version: 05.09.09.02
[ INFO] [1553070463.397931252]: Device Product ID: 0x0B07
**[ INFO] [1553070463.397985706]: Enable PointCloud: Off
[ INFO] [1553070463.398040703]: Align Depth: Off**
[ INFO] [1553070463.398100432]: Sync Mode: On
[ INFO] [1553070463.399400219]: Device Sensors: 
[ INFO] [1553070463.400089122]: Stereo Module was found.
[ INFO] [1553070463.400237230]: RGB Camera was found.
[ INFO] [1553070463.403922357]: setupPublishers...
[ INFO] [1553070463.416284782]: Expected frequency for depth = 30.00000
[ INFO] [1553070463.439415388]: Expected frequency for infra1 = 30.00000
[ INFO] [1553070463.460128902]: Expected frequency for infra2 = 30.00000
[ INFO] [1553070463.477405831]: Expected frequency for color = 30.00000
[ INFO] [1553070463.485646318]: Plugin sys_status initialized
[ INFO] [1553070463.486206105]: Plugin sys_time loaded
[ INFO] [1553070463.530140522]: setupStreams...
 20/03 17:27:43,541 WARNING [547423748096] (sensor.cpp:329) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1553070463.556539875]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553070463.557288475]: infra1 stream is enabled - width: 480, height: 270, fps: 30
[ INFO] [1553070463.557847463]: infra2 stream is enabled - width: 480, height: 270, fps: 30
[ INFO] [1553070463.560894176]: TM: Timesync mode: MAVLINK
[ INFO] [1553070463.565370334]: Plugin sys_time initialized
[ INFO] [1553070463.565663194]: Plugin trajectory loaded
[ INFO] [1553070463.580023069]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1553070463.594178497]: Plugin trajectory initialized
[ INFO] [1553070463.594490477]: Plugin vfr_hud loaded
[ INFO] [1553070463.596612823]: Plugin vfr_hud initialized
[ INFO] [1553070463.596701841]: Plugin vibration blacklisted
[ INFO] [1553070463.596942677]: Plugin vision_pose_estimate loaded
[ INFO] [1553070463.623320927]: publishStaticTransforms...
[ INFO] [1553070463.627217792]: RealSense Node Is Up!
[ INFO] [1553070463.636379033]: Plugin vision_pose_estimate initialized
[ INFO] [1553070463.636689991]: Plugin vision_speed_estimate loaded
[ INFO] [1553070463.648750603]: Plugin vision_speed_estimate initialized
[ INFO] [1553070463.649147158]: Plugin waypoint loaded
[ INFO] [1553070463.674603054]: Plugin waypoint initialized
[ INFO] [1553070463.674718101]: Plugin wheel_odometry blacklisted
[ INFO] [1553070463.675080506]: Plugin wind_estimation loaded
[ INFO] [1553070463.677476206]: Plugin wind_estimation initialized
[ INFO] [1553070463.677598319]: Autostarting mavlink via USB on PX4
[ INFO] [1553070463.677739841]: Built-in SIMD instructions: ARM NEON
[ INFO] [1553070463.677829467]: Built-in MAVLink package version: 2019.2.2
[ INFO] [1553070463.677920340]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1553070463.677985633]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1553070463.703874472]: IMU: High resolution IMU detected!
[ INFO] [1553070463.743125372]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553070464.290441085]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1553070464.314658566]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553070464.383156239]: IMU: High resolution IMU detected!
[ INFO] [1553070465.300054196]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1553070465.300201416]: VER: 1.1: Flight software:     01090080 (18f7ee2d50dc4504)
[ INFO] [1553070465.300273650]: VER: 1.1: Middleware software: 01090080 (18f7ee2d50dc4504)
[ INFO] [1553070465.300336036]: VER: 1.1: OS software:         071c00ff (ffd37a0175352697)
[ INFO] [1553070465.300414379]: VER: 1.1: Board hardware:      00000011
[ INFO] [1553070465.300464774]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1553070465.300514817]: VER: 1.1: UID:                 3337510d37353832
[ WARN] [1553070465.300631818]: CMD: Unexpected command 520, result 0
[set_RS_param-9] process has finished cleanly
log file: /home/nvidia/.ros/log/0600d7ac-4aea-11e9-9209-36006a8d354b/set_RS_param-9*.log
[ INFO] [1553070474.290864191]: HP: requesting home position
model parameter acceleration horizontal is set from  5.000000 to 5.000000 
model parameter acceleration up is set from  10.000000 to 10.000000 
model parameter landing speed is set from  0.700000 to 0.700000 
model parameter takeoff speed is set from  1.000000 to 1.500000 
model parameter velocity horizontal is set from  1.000000 to 5.000000 
model parameter velocity down is set from  10.000000 to 1.000000 
model parameter velocity up is set from  3.000000 to 3.000000 
[ INFO] [1553070479.306719960]: WP: item #0* F:3 C: 22 p: 15.000000 0.000000 0.000000 nan x: 37.512775 y: 127.032310 z: 50.000000
[ INFO] [1553070479.310987023]: WP: item #1  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 37.512615 y: 127.030586 z: 50.000000
[... cut ...]
[ INFO] [1553070479.327259934]: WP: item #4  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 37.511772 y: 127.032814 z: 50.000000
[ INFO] [1553070479.337915294]: WP: item #5  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 37.512272 y: 127.032333 z: 50.000000
[ INFO] [1553070479.343158793]: WP: item #6  F:2 C: 20 p: 0.000000 0.000000 0.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[ INFO] [1553070479.343782414]: WP: mission received
------------------------------------------------------------

mrivi, 2019-03-20:
@moonlive why do you say the local planner doesn't work? The output seems correct.

moonlive, 2019-03-20:
@mrivi 
Below is my `rostopic list` when run avoidance launch.

`nvidia@tegra-ubuntu:~/catkin_ws$ rostopic list
/adapted_waypoint
/bounding_box
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/points
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/realsense2_camera_manager/bond
/camera/realsense2_camera_manager/parameter_descriptions
/camera/realsense2_camera_manager/parameter_updates
/clicked_point
/complete_tree
/cost_image
/current_setpoint
/diagnostics
/drone
/goal_position
/ground_measurement
/histogram_image
/initial_height
/input/goal_position
/local_planner_node/parameter_descriptions
/local_planner_node/parameter_updates
/local_pointcloud
/mavlink/from
/mavlink/to
/mavros/actuator_control
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/companion_process/status
/mavros/debug_value/debug
/mavros/debug_value/debug_vector
/mavros/debug_value/named_value_float
/mavros/debug_value/named_value_int
/mavros/debug_value/send
/mavros/extended_state
/mavros/fake_gps/mocap/tf
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/home
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/rel_alt
/mavros/global_position/set_gp_origin
/mavros/gps_rtk/send_rtcm
/mavros/hil/actuator_controls
/mavros/hil/controls
/mavros/hil/gps
/mavros/hil/imu_ned
/mavros/hil/optical_flow
/mavros/hil/rc_inputs
/mavros/hil/state
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/diff_pressure
/mavros/imu/mag
/mavros/imu/static_pressure
/mavros/imu/temperature_baro
/mavros/imu/temperature_imu
/mavros/local_position/accel
/mavros/local_position/odom
/mavros/local_position/pose
/mavros/local_position/pose_cov
/mavros/local_position/velocity_body
/mavros/local_position/velocity_body_cov
/mavros/local_position/velocity_local
/mavros/log_transfer/raw/log_data
/mavros/log_transfer/raw/log_entry
/mavros/manual_control/control
/mavros/manual_control/send
/mavros/mission/reached
/mavros/mission/waypoints
/mavros/mocap/tf
/mavros/obstacle/send
/mavros/odometry/in
/mavros/odometry/out
/mavros/param/param_value
/mavros/px4flow/ground_distance
/mavros/px4flow/raw/optical_flow_rad
/mavros/px4flow/raw/send
/mavros/px4flow/temperature
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_attitude/thrust
/mavros/setpoint_position/global
/mavros/setpoint_position/local
/mavros/setpoint_raw/attitude
/mavros/setpoint_raw/global
/mavros/setpoint_raw/local
/mavros/setpoint_raw/target_attitude
/mavros/setpoint_raw/target_global
/mavros/setpoint_raw/target_local
/mavros/setpoint_velocity/cmd_vel
/mavros/setpoint_velocity/cmd_vel_unstamped
/mavros/state
/mavros/statustext/recv
/mavros/statustext/send
/mavros/target_actuator_control
/mavros/time_reference
/mavros/timesync_status
/mavros/trajectory/desired
/mavros/trajectory/generated
/mavros/trajectory/path
/mavros/vfr_hud
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_vector
/mavros/wind_estimation
/move_base_simple/goal
/original_waypoint
/path_actual
/path_adapted_waypoint
/path_waypoint
/reprojected_points
/rosout
/rosout_agg
/smoothed_waypoint
/take_off_pose
/tf
/tf_static
/tree_path
/world
`
Could i missing topic?

And, Can you show me the way how to see cloudpoint at rviz. I add depth camera but can't see images.

mrivi, 2019-03-20:
@moonlive See this post on how to connect to Rviz http://discuss.px4.io/t/obstacles-recognition/9864

baumanta, 2019-03-20:
@moonlive if you do rostopic hz /local_pointcloud and you get any frequency (about 10-20Hz usually) then your planner is running fine. If you want to visualize topics in Rviz, you need to add the topic to Rviz

moonlive, 2019-03-20:
@mrivi 
The reason i think so is because I can not see the messages that are shown when it working normally.
Like this `Planning started, using 1 cameras`.
It is according official install manual.

moonlive, 2019-03-20:
I'm appreciate help me everyone.
Hello @baumanta 
I didn't see pointcloud images but i checke data as `rostopic echo /local_pointcloud`.
It looks like frequency is very late. 
----------------------------
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00459s window: 2796
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00459s window: 2801
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00459s window: 2807
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00459s window: 2812
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00458s window: 2818
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00458s window: 2824
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00458s window: 2829
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00457s window: 2835
average rate: 5.515
        min: 0.164s max: 0.199s std dev: 0.00458s window: 2840
-----------------------------
Do you have any tip for more faster?

Thank you very much everyone again.

baumanta, 2019-03-20:
that means the planner is running correctly. If you do not see the pointcloud in Rviz you did not set up your Rviz connection properly. Please check your RViz setup again. The frequency should most likely work, but I wonder why it is so low (on TX2 you should have enough compute power to run faster)

jkflying, 2019-03-20:
You can reduce the number of nodes to get faster planning, also adjust the distance between nodes. These are in RQT reconfigure.

closed 2019-03-20 by jkflying