PX4/PX4-Avoidance #259: How to support multi-cameras in local planner

sunzj, 2019-03-21:
i find there is launch file to support 3 camera in local-planner. And there are also three "static_transform_publisher" in local_planner_sitl_3cam.launch, such as:

    
    <node pkg="tf" type="static_transform_publisher" name="tf_front_camera"
          args="0 0 0 -1.57 0 -1.57 fcu front_camera_link 10"/>
    <node pkg="tf" type="static_transform_publisher" name="tf_right_camera"
          args="0 0 0 -3.14 0 -1.57 fcu right_camera_link 10"/>
    <node pkg="tf" type="static_transform_publisher" name="tf_left_camera"
          args="0 0 0 0 0 -1.57 fcu left_camera_link 10"/>

However, i don't find code to transform point cloud from camera to fcu then to /local_origin,there is code to transform point cloud from camera to /local_origin in updatePlannerInfo(). 

So i misunderstand?! Or it just supports one camera for local-planner?

mrivi, 2019-03-21:
@sunzj That's correct. The local planner supports multiple camera. The transformation from the camera frame to the local_origin frame is done in UpdatePlannerInfo() for each pointcloud available (see this for loop https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/local_planner_node.cpp#L201). 
The transformation from camera frame to local origin is comped of multiple transformation camera_frame -> fcu -> local_origin. This is the transformation tree from the local planner simulation with 3 depth sensor

[frames.pdf](https://github.com/PX4/avoidance/files/2991074/frames.pdf)

sunzj, 2019-03-26:
@mrivi Thanks, i got it.
If there is a transformation: A->B->C->D, if need transform A->D, just pass frame_id A, and D to tf module. Then tf will handle the intermediate steps.

closed 2019-03-26 by sunzj