PX4/PX4-Avoidance #262: Any help please !

lioneleeedadad, 2019-03-25:
Error [parser.cc:689] Unable to find uri[model://iris]

mrivi, 2019-03-25:
@lioneleeedadad please follow these instructions https://github.com/PX4/avoidance#build-and-run-the-simulator

lioneleeedadad, 2019-03-25:
One more thing , is there any issue in this package or should it work normally with no errors and give me the path planning I need ?

mrivi, 2019-03-25:
@lioneleeedadad You need to adapt this line based on the location of the PX4 Firmware folder in your setup.

lioneleeedadad, 2019-03-25:
I copy in one line
. ~catkin_ws/src/Firmware/Tools/setup_gazebo.bash ~catkin_ws/src/Firmware ~catkin_ws/src/Firmware/build/px4_sitl_default

the output is
bash: ~catkin_ws/src/Firmware/Tools/setup_gazebo.bash: No such file or directory

lioneleeedadad, 2019-03-25:
@mrivi 
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models
should this line be copied as it is or I should adjust it ?

mrivi, 2019-03-25:
@lioneleeedadad I think it should be
`. ~/catkin_ws/src/Firmware/Tools/setup_gazebo.bash ~/catkin_ws/src/Firmware ~/catkin_ws/src/Firmware/build/px4_sitl_default `

It should as it is if the avoidance repository is in `~/catkin_ws/src/avoidance`

lioneleeedadad, 2019-03-26:
It worked ! THANK YOU .. YOU ARE GREAT!

closed 2019-03-26 by nicovanduijn