PX4/PX4-Avoidance #265: How can I launch local planner on gazebo ?

lioneleeedadad, 2019-03-26:


mrivi, 2019-03-26:
In the repository readme there is a section "Run the Avoidance Gazebo Simulation" with a subsection specifically on the Local Planner. Please, follow those instructions.

closed 2019-03-26 by mrivi

lioneleeedadad, 2019-03-26:
I followed these instructions however only RVIZ launches

mrivi, 2019-03-26:
@lioneleeedadad which launch file are you using? Can you please share the console output?

lioneleeedadad, 2019-03-26:
when I use roslaunch local_planner local_planner_stereo.launch 
I get:
[gazebo-6] process has died [pid 24700, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/omar/catkin_ws/src/avoidance/local_planner/../sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/omar/.ros/log/493553b8-4fef-11e9-a65b-e82a44e20f6f/gazebo-6.log].
log file: /home/omar/.ros/log/493553b8-4fef-11e9-a65b-e82a44e20f6f/gazebo-6*.log

mrivi, 2019-03-26:
@lioneleeedadad sometimes it happens that Gazebo dies. If there isn't any other error, try a couple of times and it will work.

lioneleeedadad, 2019-03-26:
Yes it worked but only RVIZ launches .. Gazebo doesn't

lioneleeedadad, 2019-03-26:
And there is no error ..

lioneleeedadad, 2019-03-26:
What can I write so that Gazebo launch ?

lioneleeedadad, 2019-03-26:
I need it very much

mrivi, 2019-03-26:
@lioneleeedadad there must be something else wrong because the launch files are correct. 
Please try `roslaunch local_planner local_planner_sitl_3cam.launch` and share the entire console output.

lioneleeedadad, 2019-03-26:
~/catkin_ws/src$ roslaunch local_planner local_planner_sitl_3cam.launch
... logging to /home/omar/.ros/log/60e833de-4ff1-11e9-a65b-e82a44e20f6f/roslaunch-omar-18781.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://omar:33473/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/camera_front/d...
 * /local_planner_node/world_name: /home/omar/catkin...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/plugin_blacklist: ['safety_area', '...
[... 205 lines cut ...]
[ INFO] [1553623379.857580957, 1.448000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1553623379.857742496, 1.448000000]: IMU: Scaled IMU message used.
[ INFO] [1553623379.858805440, 1.448000000]: IMU: High resolution IMU detected!
INFO  [mavlink] partner IP: 127.0.0.1
[ INFO] [1553623380.151964428, 1.732000000]: FCU: [logger] file: ./log/2019-03-26/18_02_59.ulg
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[ INFO] [1553623380.758603137, 2.336000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553623380.850667634, 2.428000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1553623380.861338582, 2.440000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553623380.869656601, 2.444000000]: IMU: Scaled IMU message used.
[ INFO] [1553623380.869875056, 2.448000000]: IMU: High resolution IMU detected!
[ INFO] [1553623381.857276722, 3.432000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1553623381.857361637, 3.432000000]: VER: 1.1: Flight software:     01090080 (3d271245a1052ea6)
[ INFO] [1553623381.857402137, 3.432000000]: VER: 1.1: Middleware software: 01090080 (3d271245a1052ea6)
[ INFO] [1553623381.857451761, 3.432000000]: VER: 1.1: OS software:         040400ff (ea24b543df6aab66)
[ INFO] [1553623381.857559510, 3.432000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1553623381.857671967, 3.432000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1553623381.857730550, 3.432000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
model parameter auto mode is set from  1 to 1 
model parameter takeoff speed is set from  1.000000 to 1.500000 
model parameter velocity horizontal is set from  1.000000 to 5.000000 
model parameter velocity up is set from  3.000000 to 3.000000 
[ INFO] [1553623396.050336308, 17.428000000]: WP: mission received

lioneleeedadad, 2019-03-26:
oslaunch local_planner local_planner_sitl_3cam.launch
... logging to /home/omar/.ros/log/60e833de-4ff1-11e9-a65b-e82a44e20f6f/roslaunch-omar-18781.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://omar:33473/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/camera_front/d...
 * /local_planner_node/world_name: /home/omar/catkin...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    local_planner_node (local_planner/local_planner_node)
    mavros (mavros/mavros_node)
    rqt_reconfigure (rqt_reconfigure/rqt_reconfigure)
    rviz (rviz/rviz)
    sitl_0 (px4/px4)
    spawn_model (gazebo_ros/spawn_model)
    tf_front_camera (tf/static_transform_publisher)
    tf_left_camera (tf/static_transform_publisher)
    tf_right_camera (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [18791]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 60e833de-4ff1-11e9-a65b-e82a44e20f6f
process[rosout-1]: started with pid [18804]
started core service [/rosout]
process[tf_front_camera-2]: started with pid [18829]
process[tf_right_camera-3]: started with pid [18830]
process[tf_left_camera-4]: started with pid [18837]
process[rqt_reconfigure-5]: started with pid [18850]
process[sitl_0-6]: started with pid [18886]
INFO  [px4] Creating symlink /home/omar/catkin_ws/src/Firmware/ROMFS/px4fmu_common -> /home/omar/.ros/etc
0 WARNING: setRealtimeSched failed (not run as root?)

lioneleeedadad, 2019-03-26:
roslaunch local_planner local_planner_sitl_3cam.launch
... logging to /home/omar/.ros/log/60e833de-4ff1-11e9-a65b-e82a44e20f6f/roslaunch-omar-18781.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://omar:33473/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/camera_front/d...
 * /local_planner_node/world_name: /home/omar/catkin...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
 * /mavros/plugin_blacklist: ['safety_area', '...
[... 205 lines cut ...]
[ INFO] [1553623379.857580957, 1.448000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1553623379.857742496, 1.448000000]: IMU: Scaled IMU message used.
[ INFO] [1553623379.858805440, 1.448000000]: IMU: High resolution IMU detected!
INFO  [mavlink] partner IP: 127.0.0.1
[ INFO] [1553623380.151964428, 1.732000000]: FCU: [logger] file: ./log/2019-03-26/18_02_59.ulg
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[ INFO] [1553623380.758603137, 2.336000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553623380.850667634, 2.428000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1553623380.861338582, 2.440000000]: IMU: Attitude quaternion IMU detected!
[ INFO] [1553623380.869656601, 2.444000000]: IMU: Scaled IMU message used.
[ INFO] [1553623380.869875056, 2.448000000]: IMU: High resolution IMU detected!
[ INFO] [1553623381.857276722, 3.432000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1553623381.857361637, 3.432000000]: VER: 1.1: Flight software:     01090080 (3d271245a1052ea6)
[ INFO] [1553623381.857402137, 3.432000000]: VER: 1.1: Middleware software: 01090080 (3d271245a1052ea6)
[ INFO] [1553623381.857451761, 3.432000000]: VER: 1.1: OS software:         040400ff (ea24b543df6aab66)
[ INFO] [1553623381.857559510, 3.432000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1553623381.857671967, 3.432000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1553623381.857730550, 3.432000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
model parameter auto mode is set from  1 to 1 
model parameter takeoff speed is set from  1.000000 to 1.500000 
model parameter velocity horizontal is set from  1.000000 to 5.000000 
model parameter velocity up is set from  3.000000 to 3.000000 
[ INFO] [1553623396.050336308, 17.428000000]: WP: mission received

mrivi, 2019-03-26:
@lioneleeedadad the simulation is working correctly from the output. You will see the work and vehicle in rviz. Gazebo is running headless by default. If you want the gazebo GUI you need to set [this argument](https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_sitl_mavros.launch#L6) to `true`

lioneleeedadad, 2019-03-26:
YOU ARE GREAT !! THANK YOU

optimize0107, 2019-07-08:
> @lioneleeedadad the simulation is working correctly from the output. You will see the world and vehicle in rviz. Gazebo is running headless by default. If you want the gazebo GUI you need to set [this argument](https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_sitl_mavros.launch#L6) to `true`

I can't enter that link..

nicovanduijn, 2019-07-08:
the launch files moved.
https://github.com/PX4/avoidance/blob/3acc9a55d7a57e0c34919087637ea50520bf045b/avoidance/launch/avoidance_sitl_mavros.launch#L3