PX4/PX4-Avoidance #269: What version of PX4 doese avaliable avoidance function ?

moonlive, 2019-03-28:
Hello everyone.

I am now trying to avoidance operation using PixHAWK2.1 and TX2.
I heard that the Avoidance feature was released based on V1.8.0.
But I did not check it.
Not applied to V1.8.2
Should I get it on a separate branch?

It seems to be using V1.9.0-beta1, but something seems to be unstable.
I would be grateful if you could recommend a stable and avoidance version.

mrivi, 2019-03-28:
@moonlive - hi, Firmware version 1.8.x doesn't support obstacle avoidance. If you want to use a release, V1.9.0-beta1 is the one. Otherwise you need to use master. 
As far as the avoidance, there is only one stable release 0.1.0 .

moonlive, 2019-03-28:
Ok I got it. 
Thank you very much

closed 2019-03-28 by moonlive