PX4/PX4-Avoidance #271: Please I'm stucked .. would please anyone tell me the available online path planning feature in this package ?

lioneleeedadad, 2019-03-28:


mhkabir, 2019-03-29:
Please stop opening duplicate issues for discussion questions.

Yes, this package implements methods for online planning, using 3DVFH+ for local avoidance and RRT on Octomaps for global avoidance. The local planner is currently well supported.

closed 2019-03-29 by mhkabir

lioneleeedadad, 2019-03-29:
@mhkabir I apologiza about that. But would you please tell me how can I switch to online mode ?

nicovanduijn, 2019-03-29:
Frankly, your question is a bit like asking on the NASA forum whether you should use slot drive or cross screwdrivers to build your space shuttle.

Please start by reading the readme, following the instructions for installation, probably best in simulation first, and maybe reading up about the basics of path planning.

There is no offline planner in this repository, so clearly you got some of the fundamentals wrong.

lioneleeedadad, 2019-03-29:
@nicovanduijn @mhkabir Yes thats great but how can I add dynamic obstacles

lioneleeedadad, 2019-04-03:
@baumanta @mhkabir kabir mohammed are you sure it is 3DVFH+ not 3DVFH* as I saw a topic written by @baumanta  "The local_planner uses a 3DVFH* algorithm. It is to some extent a combination of the 3DVFH concept and the VFH* concept plus a few custom modifications to make the algorithm suitable for real world situations. I attach papers to the two mentioned algorithms in case you would like to read up on it." so is it 3DVFH+ or 3DVFH* ?!