PX4/PX4-Avoidance #276: Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/build.make:62: recipe for target 'Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/pga460.cpp.o' failed make[2]: *** [Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/pga460.cpp.o] Error 1 CMakeFiles/Makefile2:7339: recipe for target 'Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/all' failed

lioneleeedadad, 2019-03-30:


Jaeyoung-Lim, 2019-03-30:
@lioneleeedadad What is the question? I don't see anything in the description

lioneleeedadad, 2019-03-30:
I can't do catkin_make .. and this error is appearing

lioneleeedadad, 2019-03-30:
there is an error in Firmware !

Jaeyoung-Lim, 2019-03-30:
If you are doing catkin_make you clearly are not following the setup process on the readme of this repo.

Could you try following the readme?

In order for me to help, you need to "describe" your problem in the description. as detailed as possible- what you are trying to do, how you tried and what is not working.

lioneleeedadad, 2019-03-30:
I followed the setup process how I now cloned the rqt reconfigure and other packages related to rqt and then I'm trying to do that catkin_make and I'm recieving output 
cc1plus: error: assuming signed overflow does not occur when assuming that (X + c) < X is always false [-Werror=strict-overflow]
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/build.make:62: recipe for target 'Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/pga460.cpp.o' failed
make[2]: *** [Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/pga460.cpp.o] Error 1
CMakeFiles/Makefile2:10238: recipe for target 'Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/all' failed
make[1]: *** [Firmware/src/drivers/distance_sensor/pga460/CMakeFiles/drivers__pga460.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

lioneleeedadad, 2019-03-30:
Also this issue is related to PX4 avoidance package thats why I'm asking here .. So please help me :)

Jaeyoung-Lim, 2019-03-30:
This is much more clear.

So to make this clear. This is not an issue with the avoidane package itself, but a problem with your installation.

I see a few problems from your log:
- where is your firmware located? It shouldn't be in the `catkin_ws` Please follow the installation instructions precisely on the devguide: http://dev.px4.io/en/setup/dev_env_linux.html
- You shouldn't be using `catkin_make` as the avoidance package builds on `catkin build` You cannot use both in the same workspace. You need to clean the workspace before you do any builds
- I am curious if you were able to run any ros node with your system? If not, to verify everything is fine try following one of the ROS examples so that at least you are sure everything is alright with ROS-and rqt reconfigure

lioneleeedadad, 2019-03-30:
My Firmware is in catkin_ws/src/

Jaeyoung-Lim, 2019-03-30:
@lioneleeedadad Exactly in shouldn't be there

lioneleeedadad, 2019-03-30:
I will try to add it in its right place and tell you what happened

mhkabir, 2019-03-30:
@lioneleeedadad Please use discuss.px4.io for help instead of opening new issues. We will close and lock issues which do not include information and do not follow standard guidelines.

closed 2019-03-30 by mhkabir