PX4/PX4-Avoidance #277: RRT* PATH PLANNING

mechaliomar, 2019-04-01:
Hello, Because the team is currently working on path planning , why they don't use for example RRT* as local planner, it's asymptotically optimal. I had used it on real experiments on DJI M100 drone for path planning with pre-loaded obstacles positions.
And different study show that RRT is really an efficient algorithm for path planning in complex environment.
Best regards
Mecahli Omar
mechaliomar@gmail.com

Jaeyoung-Lim, 2019-04-01:
RRT* is a sampling based planner which is used for finding globally optimal paths. As the RRT* samples in the workspace, this means that you need to create a map in order to be able to use RRT*

I agree that RRT* could be an attractive planner to use, but `local_planner` is designed so that we don't need to create a map and plan trajectories faster than sampling based algorithms with much lower computational costs.

The `global_planner` might be closer to what you are describing.

jkflying, 2019-04-01:
There would also need to be some accounting for the kinematics of the drone, which affects the paths available, otherwise it wouldn't work in more complex environments while at practical speeds. Something like this might be feasible once a map is constructed: https://arxiv.org/pdf/1205.5088.pdf

closed 2019-04-01 by jkflying

mechaliomar, 2019-04-02:
I'm always wondering about how to avoid an obstacle which come suddenly in front of the UAV. I used RRT* with preloaded obstacles , but how to avoid them in real time?

Jaeyoung-Lim, 2019-04-02:
@mechaliomar There is a large body of literature on incremental replanning in unknown environments. It is not answer. Something like this might be interesting for you: https://arxiv.org/abs/1710.00604

mechaliomar, 2019-04-03:
@JaeyoungLIm Thnaks a lot for that paper, it's really useful. I'm reading
it to get some information. XIEXIE

mechaliomar, 2019-04-03:
@JaeyoungLIm I also took a look at :
https://github.com/ethz-asl/mav_voxblox_planning
What do you think about this Github material?