PX4/PX4-Avoidance #283: global_planner and local_planner

lioneleeedadad, 2019-04-03:
Dear people 
I have a question
what kind of path planning is used for global path planning and local path planning ?

baumanta, 2019-04-03:
this is not an issue

closed 2019-04-03 by baumanta

lioneleeedadad, 2019-04-03:
@baumanta hello baumanta yes but I saw the same question in the closed issues and there was other questions .. also this question will help other users

lioneleeedadad, 2019-04-03:
@baumanta also I intend to ask here as it takes too much time to get a reply on px4 discussion

Jaeyoung-Lim, 2019-04-03:
@lioneleeedadad You will not get an answer here. And as you said, this is a duplicate question, so please refer to the answers that already exist. 

Please use discuss to ask questions and use this thread for issues only.