PX4/PX4-Avoidance #290: local_planner_node_main syntax error

lioneleeedadad, 2019-04-05:
rosrun local_planner local_planner_node_main.cpp
/home/omar/catkin_ws/src/avoidance/local_planner/src/nodes/local_planner_node_main.cpp: line 8: syntax error near unexpected token `('
/home/omar/catkin_ws/src/avoidance/local_planner/src/nodes/local_planner_node_main.cpp: line 8: `int main(int argc, char** argv) {'
this error is in the node when I try to run it alone

Jaeyoung-Lim, 2019-04-05:
The rosrun command should be 
```
rosrun <package_name> <node_name>
```
Therefore, your ron run command is not correct. There are proper instructions in the README on how to launch the local_planner node

lioneleeedadad, 2019-04-05:
I only need to run this node

lioneleeedadad, 2019-04-05:
rosrun local_planner local_planner_node_main.cpp 
I need to run the local_planner_node_main . to modify on it

Jaeyoung-Lim, 2019-04-05:
`local_planner_node_main.cpp` is not a node. It is the source code for the node.

Can you explain why you need to 'only' run the main function?

lioneleeedadad, 2019-04-05:
forsure .. I need to connect the local path planner to a publisher node . So I thought that the node which should be a subscriber is the local_planner_node_main

lioneleeedadad, 2019-04-05:
What is the node that I could use to subscribe to a topic ?

Jaeyoung-Lim, 2019-04-05:
A node cannot be a subscriber. If you simply want to control the drone using ROS topics, you can use [mavros](https://github.com/mavlink/mavros)

lioneleeedadad, 2019-04-05:
why can't it subscribe to a topic ?

Jaeyoung-Lim, 2019-04-05:
The node itself subscribes to topics. If you want the `local_planner` to subscribe to a new topic, you need to modify the source code

I think you are confused on how ROS works in general. Following a ROS tutorial and writing a simple node will let you better understand what a node, subscriber is and how it works with the local planner

lioneleeedadad, 2019-04-05:
No i just need one node to subscribe to this topic not the hole local_planner .

lioneleeedadad, 2019-04-05:
I'm thinking of local_planner_node to subscribe to that topic . I just need the local planner node to print hello world

Jaeyoung-Lim, 2019-04-05:
@lioneleeedadad In that case writing a new node for your own porpose would be a better solution

lioneleeedadad, 2019-04-05:
I did them all :D

lioneleeedadad, 2019-04-05:
and I know how to do node

lioneleeedadad, 2019-04-05:
I just need to publish on a local path planning node to control it either to switch it on or off

Jaeyoung-Lim, 2019-04-05:
@lioneleeedadad Then you need to modify the source code of the local_planner

lioneleeedadad, 2019-04-05:
which one is the source code ?

lioneleeedadad, 2019-04-05:
all what I'm looking forward to is a publisher which publisher messages to say chatter topic and this message is either 0 or 1 if its 0 local path planning will be turned off and a teleop node will work if it is 1 local path path planning will work

lioneleeedadad, 2019-04-05:
and this 0 or 1 is sensor signals

Jaeyoung-Lim, 2019-04-05:
@lioneleeedadad You need to write your own package for that. http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29 this tutorial can be useful for you

lioneleeedadad, 2019-04-05:
I want to utilize this package to do that

Jaeyoung-Lim, 2019-04-05:
As I said, you need to reprogram this package

lioneleeedadad, 2019-04-05:
yes I'm working on that

lioneleeedadad, 2019-04-05:
however which source files should I edit ?

Jaeyoung-Lim, 2019-04-05:
https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/local_planner_node.cpp#L52 This is where the subscribers are initialized in the local_planner

lioneleeedadad, 2019-04-05:
Okay then I need to write the subscriber node then to initialize it using this node you just mentioned ?

mhkabir, 2019-04-06:
Please move to Discuss.

closed 2019-04-06 by mhkabir