PX4/PX4-Avoidance #294: Avoidance system not available!

trgiman, 2019-04-08:
Hello,

I am able to run 

roslaunch local_planner local_planner_stereo.launch an plann path for UAV

but in case of 

1.roslaunch global_planner global_planner_sitl_mavros.launch I have got following error

WARN  [commander] Avoidance system not available!
[ERROR] [1554717383.081456403, 886.980000000]: FCU: Avoidance system not available!

same error is for: 

2.roslaunch local_planner local_planner_sitl_mavros.launch 

In case of 1. Gazebo world,iris model,PX4 simulation is loaded but since Avoidance system is not available I can not arm and take-off and plan the path?  However in RVIZ - point cloud, octomap,everything seems to work correctly Any ideas?

a)My gazebo ros kinetic version is 9.6 (updated from 7) and checked with sudo apt-get install ros-kinetic-gazebo9-* that no packages are missing in order to make gz plugins to work properly.

b)PX4 version is v1.9.0-beta2

Thank you for recommendations.

mrivi, 2019-04-08:
Hi @trgiman , `local_planner local_planner_sitl_mavros` it's an intermediate launch file that's why it doesn't work. In order to launch the local_planner you need to use `local_planner_depth-camera`, `local_planner_sitl_3cam` or `local_planner_stereo`. More details on the difference between these simulation is provided in this package README. 
As for the global planner, you will need to downgrade to v1.8 stable release and use it only in offboard mode. Sorry about that. We will try to have it working on 1.9 as soon as possible.

trgiman, 2019-04-08:
Hi @mrivi,

Thanks a lot for info. Actually my first attempt was with v1.8. But it did not have px4.launch file and compilation had to be done trough make posix_default_sitl (option from tutorial not available). After modifying launch file etc. accordingly, result was unfortunately, that ros was reporting multiple nodes named /gazebo. 

So I tried to add px4.launch from git, but in that case Gazebo models etc are not loaded. Now I am running local_planner_depth-camera which runs smoothly (PX4 v1.9.0-beta2)

1.8v is not working for me in regards of Global Planner.

mrivi, 2019-04-08:
@trgiman The avoidance should work with 1.8 (both local and global planner). Can you share the error output? Thanks

trgiman, 2019-04-08:
@mrivi .Sure. Basically I just followed the the Readme tutorial.

1. cloned current PX4 and updated submodules
2. git checkout v1.8.0
3. make list_config_targets  - result is px4_sitl_default is not available so used make posix_sitl_default
4.Configuration of PATHes in respect of my Firmware dir and posix_sitl_default option 
5.Launching .roslaunch global_planner global_planner_sitl_mavros.launch

Error 1:
while processing /home/trgiman/pplanningPX4/src/avoidance/global_planner/launch/mavros_sitl.launch:
while processing /home/trgiman/PX4/Firmware/launch/px4.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/trgiman/PX4/Firmware/launch/px4.launch'

Now option 1 add px4.launch file or  option 2 edit  mavros_sitl.launch. Tried both as mentioned above. Or maybe I am doing something essentially wrong?

mrivi, 2019-04-08:
I think for 1.8 to work you need to revert this change https://github.com/PX4/avoidance/pull/98/files

trgiman, 2019-04-08:
Hi @mrivi I did reverted to commit, that was recommended by you. Everything went well but unfortunately this time I get following gazebo error:

Aborted (core dumped)
[gazebo-7] process has died [pid 5303, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/trgiman/pplanningPX4/src/avoidance/global_planner/../sim/worlds/test_city_2.world __name:=gazebo __log:=/home/trgiman/.ros/log/66c5d584-5a34-11e9-aaed-0c9d928302ea/gazebo-7.log].
log file: /home/trgiman/.ros/log/66c5d584-5a34-11e9-aaed-0c9d928302ea/gazebo-7*.log

FYI. I did compilation of PX4 and Avoidance from clean git clone etc. Log above is non existent...in which launch file I could apply verbose parameter for gazebo? Thanks.

mrivi, 2019-04-08:
@trgiman Those seems some warning that are always there. I don't see any error that would make the simulation fail. Is there an error?

I would advise you to first make sure that your setup is working properly with the Firmware master and avoidance master. Realease 1.8 is quite old and it doesn't support all the feature of avoidance (e.g. no OA with mission mode). Is there any specific reason why you want to use a release?

Furthemore, some issues might be caused by having Gazebo9 with ROS Kinetic. I think no one in the dev team has that setup. We recommend ROS Kinetic and Gazebo 7.

mrivi, 2019-04-08:
@trgiman By the way to use the global planner with release 1.9 or the Firmware master, you can set `COM_OBS_AVOID` to 0 and run the planner in Offboard mode. You need to reboot after setting the parameters such that the preflight checks don't fail.

trgiman, 2019-04-09:
**Hi @mrivi perfect. I set COM_OBS_AVOID to 0 and global planner works like a charm with 1.9, gazebo 9.6 on kinetic**.Hmm. I just do not get why we have to set COM_OBS_AVOID to 0 instead of leaving it to 1? It is PX4 flag to enable obs avoidance? 

Do you have got any explanation please? Initially I although tough that I can change the COM_OBS_AVOID by default in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/1015_iris_obs_avoid...but it did not chage the parameter so I simply used QGroundControll to change it.

Anyway thanks for help.

mrivi, 2019-04-09:
@trgiman COM_OBS_AVOID enables running the obstacle avoidance in MISSION mode. It isn't necessary to run it in OFFBOARD mode.

There is a bug in the Firmware with loading vehicle specific paramters from ROS. If you delete `~/.ros/eeprom/parameters` when changing vehicle type you will see that the parameters get set correctly

trgiman, 2019-04-09:
@mrivi thanks. It works I think we can close this issue.

closed 2019-04-09 by trgiman