PX4/PX4-Avoidance #295: No drone model shown, after launch "global_planner_sitl_mavros"

ElderWanng, 2019-04-08:
I run command `roslaunch global_planner global_planner_sitl_mavros.launch`. And after gazebo show black screen and "preparing world" for about 3 minus ( which I think is abnormal too), finally the buildings appear.
But I can't find the drone in front of the house...
And, In the list `world`, I unfold the `models` option and I find `Iris` in the list. I click the `follow` function, the camera just flies away... the screen turn into grey...
Then I open a new command window and run `rostopic echo /mavros/state`, I find the `connected` option is `false`...

I need some help. THX!

Jaeyoung-Lim, 2019-04-08:
@ElderWanng Have you tried adding the model path?

You can do it by the following
```
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models
```

ElderWanng, 2019-04-08:
> @ElderWanng Have you tried adding the model path?
> 
> You can do it by the following
> 
> ```
> export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models
> ```

I know this path is of importance and I checked this path for many times before ask this issue.
Unluckily, this path is working well: Open a Gazebo directly and open the `insert` section, I can find this path are shown in the UI of `gazebo`. I can drag some model such as `'Distance Sensor` into the screen. However when I drag any one of the `3DR Iris` models, BOOOM!!! the Gazebo program crashes!!!! ( the window of gazebo turns dark and Ubuntu tell me this program crashed )

My ROS is kinetic, my gazebo is 9.8.
 
But why the models can't be parsed on my computer...

Jaeyoung-Lim, 2019-04-08:
@ElderWanng You should not try and insert the IRIS model using the gui 'insert' tab. Does this mean you cannot see the iris by launching the launch file?

Do you see any errors in the log?

baumanta, 2019-04-08:
As you use ROS kinetic, I assume you are running under Ubuntu 16.04. For that OS gazebo version 7 is the supported one. Maybe gazebo 9 is causing your problem.

http://gazebosim.org/tutorials/?tut=ros_wrapper_versions

ElderWanng, 2019-04-08:
Thanks to all the people who helped me!
Finally I found all I need to do is closing the VPN ( because I'm in the country where can't use google... XD 
Thanks!

ElderWanng, 2019-04-08:
What's more, remember to Download gazebo models first.
I reinstalled my ros and forget to set up it, which cost me I have to spend 3 more hours to find this mistake....

those command would be helpful

>   cd ~/.gazebo/
>   mkdir -p models
>   cd ~/.gazebo/models/
>   wget http://file.ncnynl.com/ros/gazebo_models.txt
>   wget -i gazebo_models.txt
>   ls model.tar.g* | xargs -n1 tar xzvf

closed 2019-04-08 by ElderWanng