PX4/PX4-Avoidance #296: error while starting global_planner_sitl_mavros.launch - px4 connection doesn't work

hsu-ret, 2019-04-08:
Hey guys 
currently I started to work with px4 and ros - and if I start a px4_sitl with gazebo and a separate ROS node it works fine (see: https://dev.px4.io/en/ros/mavros_offboard.html)

While searching for additional code examples I found this px4 add-on and it looks really cool but I can't  start it :'( 
I **was** able to spawn an the world-file without the drone (just the city) but right about now gazebo stays black and can't even load the city anymore. Other gazebo simulations run still fine - so my gazebo is ok. (I can even start the px4_sitl with the iris quad which seemed to not be found by the global_planner)

As far as I can determine - the problems occurs here - it's where the PX4 symbol would appear:
> INFO  [px4] Creating symlink /home/mark/PX_4_GIT/catkin_ws/src/Firmware/ROMFS/px4fmu_common -> /home/mark/.ros/etc
> 0 WARNING: setRealtimeSched failed (not run as root?)
> 
> px4 starting.
> 
> INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
> Error: Unknown model 'iris_obs_avoid'
> ERROR [px4] Startup script returned with return value: 256

when the model of the drone is loaded it seems to time out
> [INFO] [1554728601.062677, 0.000000]: Calling service /gazebo/spawn_sdf_model
> [INFO] [1554728601.094324, 786.964000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name iris
> [spawn_model-9] process has finished cleanly

at least I was not able to spawn a world with a drone - currently my gazebo stays completely black and does nothing (at least for about 5-7 mins afterwards I shut it down)

While searching for a fix - I stumbled on quite a lot of issues which seemed to be similar by setting the GAZEBO_MODEL_PATH... I set these in my bash.rc file (so they will be exported when I open a console)

```
source /opt/ros/kinetic/setup.bash
source /home/mark/catkin_ws/devel/setup.bash
source /home/mark/PX_4_GIT/catkin_ws/devel/setup.bash

echo "ROS and catkin_ws setup Bash is loaded"
echo "--------------------------------------"

export GAZEBO_RESOURCE_PATH=/usr/share/gazebo-7
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:~/PX_4_GIT/Firmware/build/px4_sitl_default/build_gazebo
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/PX_4_GIT/Firmware/Tools/sitl_gazebo/models
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:~/PX_4_GIT/Firmware/build/px4_sitl_default/build_gazebo
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/PX_4_GIT/catkin_ws/src/avoidance/sim/models
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/PX_4_GIT/Firmware
```
Yeah I don't use:
`. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default`
Mostly because of the echo and I found one solved issue which said - that doing this directly helped him... but not for me I guess.

As you can see I have 2 catkin_ws and my workspace for the px4 is inside an additional folder. I don't know if this causes the error - but mavros/mavlink would cause issues if I ran them in my old catkin_ws so I created the new one. Usually it shouldn't be a problem because I adapted the correct PATH's and the rospackage find themselves - but maybe inside the .launch files is some hardcoded reference? 

If it helps you: I'm using ubuntu 16.04 and ros-kinetic with gazebo7

I hope someone can help me out with my issue!

mrivi, 2019-04-08:
It seems that it doesn't find the vehicle model `Error: Unknown model 'iris_obs_avoid'` . This something that recently got changes in the PX4 Firmware. Either you can update the Firmware or you can revert the vehicle type to `iris`. The only difference between the two is that `COM_OBS_AVOID` is not enabled by default with the `iris` model.

hsu-ret, 2019-04-08:
> It seems that it doesn't find the vehicle model `Error: Unknown model 'iris_obs_avoid'` . This something that recently got changes in the PX4 Firmware. Either you can update the Firmware or you can revert the vehicle type to `iris`. The only difference between the two is that `COM_OBS_AVOID` is not enabled by default with the `iris` model.oh yeah thx a lot that was a simpler fix then expected.

if anyone has the same problem - this is the path where you need to change the model description:

` ~/catkin_ws/src/avoidance/global_planner/launch/mavros_sitl.launch`

it is the second code block:
```
    <include file="$(find px4)/launch/px4.launch">
      <arg name="vehicle" default="iris"/> 
    </include>
```

closed 2019-04-08 by hsu-ret