PX4/PX4-Avoidance #301: Meaning of the mean_point in getDirectionFromTree()

sunzj, 2019-04-10:
@mrivi @baumanta 

I am not understand the meaning of  mean_point in getDirectionFromTree()
https://github.com/PX4/avoidance/blob/e723f87c18723954c9a8f8ec8974581b6cc89bcb/local_planner/src/nodes/planner_functions.cpp#L513-L515

1.As the cos_alpha calculate by distances[wp_idx + 1],distances[wp_idx] and node_distance 
https://github.com/PX4/avoidance/blob/e723f87c18723954c9a8f8ec8974581b6cc89bcb/local_planner/src/nodes/planner_functions.cpp#L506-L509
the index of distances  is [wp_idx + 1], [wp_idx]
but mean_points use path_node_positions_extended[wp_idx - 1] and path_node_positions_extended[wp_idx],the index of  path_node_positions_extended is [wp_idx - 1],[wp_idx]. 

Could you share why use different index there?

2. Both path_node_positions_extended[wp_idx - 1] and path_node_positions_extended[wp_idx] are vectors, add them in such way, it's a little strange.

As i known, if we have vector trans_vector = path_node_positions_extended[wp_idx-1] - path_node_positions_extended[wp_idx] , we may get the mean_points with
mean_point  = path_node_positions_extended[wp_idx] + alpha*trans_vector.

baumanta, 2019-04-10:
maybe this graphic helps visualizing what we do. As we want to travel forward on the path, we need to set the waypoint between the next two nodes. Otherwise we would just pull the drone towards the projection on the path.
[image]

sunzj, 2019-04-11:
@baumanta Thanks for your explanation. 

For question 1, so actually the waypoint is a forecast way-point between next path nodes.

For 2.   If expand and simplify mean_point = path_node_positions_extended[wp_idx] + alpha*trans_vector, alpha = (1-p), the result is small as 3.23.

However, if cos(alpha) less than 0,and Li and p is negative.  How about such case?

baumanta, 2019-04-11:
@sunzj  there is no such case. We first determine between which nodes the drone is currently located (in this example n3 and n4). Then, we calculate p as the fraction traveled on the direct line (n3, n4). So by definition p is in [0,1].

sunzj, 2019-04-11:
@baumanta 
such case may happens, for example.

[image]

baumanta, 2019-04-11:
@sunzj  oh yes I think you are right! good catch! I think we should clamp `p` to be in [0,1] then everything should work as expected

altshuler, 2019-05-02:
@baumanta 
I think (3.21) should be   **li=di*cos⁡(δ)**   instead  **li=Lnodes*cos⁡(δ)**

sunzj, 2019-05-05:
@altshuler the code is di*cos⁡(δ) https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/planner_functions.cpp#L517
It could be a typo on the paper.

altshuler, 2019-05-06:
Ok, thanks. That exactly what I mean.

baumanta, 2019-07-30:
this function has been completely rewritten and this corner case was fixed.

closed 2019-07-30 by baumanta