PX4/PX4-Avoidance #303: Mission mode avoidance on IntelAero with local planner

ank27, 2019-04-11:
Hi,
I am trying to run avoidance module with intel aero compute board + IntelAero realsense D435 camera running as companion system with px4 firmware 1.9.0, but while running it in mission mode with COM_OBS_AVOID=1, it reject the mission and did not go in mission mode (it didn't even arm as well).

Below are the configurations i am using:
PX4 1.9.0 build 
with COM_OBS_AVOID=1
Intel realsense camera D435 
and configurations mentioned here https://github.com/PX4/avoidance#local-planner

While running on normal mission mode with COM_OBS_AVOID=0, mission mode completes well, but as soon as COM_OBS_AVOID=1 parameter sets, it doesn't take mission. 

i have tried running the local_planner node as well alongside COM_OBS_AVOID=1, but still no luck. i have also tried to change COM_OBS_AVOID values after takeoff but it rejected the mission mode there as well.

I am able to get the data on camera pointcloud as well data on mavros topics.

Can anyone help me with how to proceed with the running it on hardware so it gets the mission with local_planner.

mrivi, 2019-04-12:
Hi @ank27, you please share the flight controller logs of both successful and rejected missions? Thanks

ank27, 2019-04-12:
Hi @mrivi , attached are the successful flight logs and unsuccessful flight logs.
Successful flight logs contains mission mode behaviour with COM_OBS_AVOID=0
and unsuccessful is when COM_OBS_AVOID=1, we did multiple tests with avoidance system running, because it was not arming initially.
then we did com_obs_avoid=0 we takeoff and then applied com_obs_avoid=1 but then it went to hold mode.

[flight_logs_avoidance.zip](https://github.com/PX4/avoidance/files/3072492/flight_logs_avoidance.zip)

Jaeyoung-Lim, 2019-04-12:
@ank27 It would make it much easier if you share the logs through [flight review ](https://review.px4.io/)

ank27, 2019-04-12:
@Jaeyoung-Lim here it is :

Successful flight log:
1)
https://review.px4.io/plot_app?log=5dec91be-4dd5-48b2-80a6-1e5e62e82e05
2)
https://review.px4.io/plot_app?log=fa516518-030d-4e6b-8766-b3f92e80ffb5

Unsuccessful flight log:
1)
https://review.px4.io/plot_app?log=c3c1f577-fc01-420d-a89d-95c481634ec7
2)
https://review.px4.io/plot_app?log=ae40cd1f-c7a7-46ee-a127-d64fa71ef78e
3)
https://review.px4.io/plot_app?log=984751e0-94be-404a-a04f-7dee33ff7254

mrivi, 2019-04-12:
@ank27 the avoidance setpoints are not correctly getting to the Flight Controller.
You can see the warning in the log `Obstacle Avoidance system failed, loitering`

Please make sure that you're receiving the mavlink message TRAJECTORY_REPRESENTATION_WAYPOINTS in the Firmware. You can change the SDLOG_PROFILE paramter to also log avoidance and computer vision messages. This way you can see from the log the vehicle_trajectory_waypoint message.

ank27, 2019-04-12:
@mrivi we were getting "Obstacle avoidance system regained" messages, we were getting loitering when we were in mission but then changed com_obs_avoid=1.

how can i make sure that it will always have avoidance system retained, rather than keep on failed and regained over and over again?

mrivi, 2019-04-12:
@ank27 it means that the link between the companion computer and the FCU has a lot of packets drops and it isn't safe to fly. Try to switch to a lighter mavlink mode e.g. MAVLINK_MODE_EXTVISION .

ank27, 2019-04-13:
@mrivi Sure, I'll try with lighter mavlink node, do I also have to change the baud rate from 57600 to 921600, as it's keep showing 'not enough bandwidth for log streaming'?

Jaeyoung-Lim, 2019-04-13:
@ank27 57600 is definitely too slow for what you are trying to do. 921600 is a more appropriate baudrate for a serial connection.

ank27, 2019-04-24:
@Jaeyoung-Lim i have changed the baud rate to 921600, but still got "not enough bandwidth for log streaming".
@mrivi i can't see `MAVLINK_MODE_EXTVISION ` but i have these modes attached in the pic.
i have set it to `minimal` for now, is that good?

[image]

mrivi, 2019-04-24:
@ank27 no, the mavlink messages required to run the avoidance in mission mode are streamed only with mavlink modes normal, onboard and extvision. Both normal and onboard are quite heavy because they stream a lot of topics that aren't necessary for the avoidance. 
Which Firmware version are you running?

ank27, 2019-04-24:
@mrivi i am using firmware version 1.9.0

ank27, 2019-04-24:
@mrivi i have changed the mavlink mode with manual entry "8",which gives "extvision" in mavlink status.
but still getting `not enough bandwidth for log streaming`.

please check attached screenshot
[image]

[image]

ank27, 2019-04-24:
thank you @jkflying, now i am using 92160 B/s, which is maximum i can use.
is it good?

jkflying, 2019-04-24:
I would suggest more like 80000 B/s, to give some safety margin.

ank27, 2019-04-24:
@jkflying seems to work now. thank you
we will try it in mission mode . :)

jkflying, 2019-04-24:
You shouldn't be seeing these, try turning off your SDLOG_PROFILE items to see if it is bandwidth related, otherwise I would say maybe you have some reliability issues with your serial link - check the open PR on mavlink-router for using low latency serial if you are using a USB-serial adapter.

closed 2019-04-24 by jkflying

Cristian-wp, 2021-06-18:
Hi I have your same problem but in mission mode. How did you solve this?