PX4/PX4-Avoidance #309: Out door test fail after integrate "Change recursion strategy #286" with Realsense R200

sunzj, 2019-04-16:
@baumanta 

After integrate https://github.com/PX4/avoidance/pull/286, out door test fail.

However, if use the previous  PR "Remove legacy code #284" https://github.com/PX4/avoidance/pull/284, it works well.

So the result is, out door test. PR 284 work well, PR 284 + PR 286 fail.

Since i use realsense R200, the cloudpoints may not good enough. Did you verify on realsense R200 or other cameras?

And i find the final cloud store more than 20 seconds cloud points, it may have too much noise in it for R200.

jkflying, 2019-04-16:
I think the problem here is the duration of the history, combined with the big clipping radius, gives lots of noise. Have you tried reducing these two? It should be easy to change the defaults in the rqt_reconfigure.

We are testing with the D435, I'm not sure of the noise/performance compared to the R200, but maybe you can load the 'accurate' profile and it will have less noise. This is what we did for the D435.

jkflying, 2019-04-17:
There is also the possibility that previously any noise in the camera FOV would be deleted and replaced with the new camera data, since it was a histogram. This no longer happens, since it is a point cloud. We could definitely add this back.

sunzj, 2019-04-17:
i don't have chance to do outdoor test now, i'll share result once test outdoor with reducing "duration of the history" and "clipping radius"

sunzj, 2019-04-25:
Test outdoor, set the max_age to 20 frames, box_radius_ to 6,the issue still exist.

jkflying, 2019-04-25:
Hi, did PR310 fix the issue?

sunzj, 2019-04-25:
@jkflying no,PR310 doesn't work, but the result was a little better, the drone followed the "mission path" a short time, then stopped, and actually there wasn't any obstacle.

and i also tried PR 286 + PR310 + noise-supression-rebased, set max_age to 8, min_num_points_per_cell_ to 12, box_radius_ to 6, the result was much better,it stopped at the front of obstacle a long time, yawed left/right , finally found a path.

nicovanduijn, 2019-04-25:
Do you have a flight log or a rosbag or something? I'm curious as to what may be the root cause

sunzj, 2019-04-25:
@jkflying @nicovanduijn  finally it works, decrease goal_cost_param_ to 1.0,   PR 286 + PR310 + noise-supression-rebased. max_point_age_s_:8, min_num_points_per_cell_:12.

It stopped at the front of the obstacle, yawed left/right should causes by the cost of change direction is too high.

jkflying, 2019-05-10:
With R435 we have increased the `min_num_points_per_cell_` all the way to 25. Could you check if maybe higher values fixes this for you? Note, there was also an issue with offtrack, it will help if you rebase as well.

sunzj, 2019-05-24:
We don't have D435, and we set `min_num_points_per_cell_ to` 15 for R200, it works well.
The issue is fixed after #325

closed 2019-05-24 by sunzj