PX4/PX4-Avoidance #332: 3D map, global_planner, pixhawk

acpl00, 2019-04-30:
Dear people.

After struggling with rtabmap_ros to make an accurate 3D map and after reading some of the developers comments on my previous posts about having a good position source that I can be used with global_planner to construct a 3D map that can be used for path planning.

I do have the following:
* A ZED stereo camera which delivers a zed odom information.
* 3DR GPS information from a PixHawk mini.
* I also have a PX4FLOW sensor.

I have the followoing questions:

1.- Can I used the previous information, I mean zed odom, gps, px4flow to build up a 3D map using global_planner?  If yes how can I do it?

2.- Before I use it into real drone can I simulate it and how?

3.- Is it possible to fuse px4flow with gps info?

I am looking forward to your replay.

Thank you.

mrivi, 2019-04-30:
@acpl00 
1. the global planner builds a map with the octomap package based on an input pointcloud. The source of such pointcloud doesn't matter but it needs a pointcloud.
2. yes, there is a working simulation. You can follow the instructions on the readMe of this package. 
3. flow fusion with gps is handled by the Firmware. For more information https://docs.px4.io/en/sensor/optical_flow.html

acpl00, 2019-04-30:
Thank you for the answer.

I have installed avoidance and when I run the command `roslaunch global planner_sitl_mavros.launch` i get the error `core dumped` and I figure it out that when I comment out the lines `
    <node name="spawn_model" pkg="gazebo_ros" type="spawn_model" output="screen"
          args="-sdf -database $(arg model) -model iris">
    </node>`

The launching file `roslaunch global planner_sitl_mavros.launch` does not crash.

Do you have any clue about this error?

Thank you

Jaeyoung-Lim, 2019-04-30:
Have you done
```
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/sim/models
```

acpl00, 2019-05-01:
Hi Jaeyoung-Lim

Thank you for your answer :)

Yea I did exported the `GAZEBO_MODEL_PATH` as you mentioned.  I had to download the avoidance package and compile it again and now I can see the drone :)

But when I run the command  `rosrun mavros mavsys mode -c OFFBOARD` the drone does not start flying instead I get the following:

`$ rosrun mavros mavsys mode -c OFFBOARD`
`Mode changed`

`[ WARN] : CMD: Unexpected command 176, result 0`

Do you have any clue about this,

Thank you

Jaeyoung-Lim, 2019-05-01:
It means that the mode has successfully changed. If you arm the drone it will fly.

acpl00, 2019-05-02:
Hi Jaeyoung-Lim

Thank you for your answer :)

I can now run and plan the goal point in RVIZ
`$roslaunch global planner_sitl_mavros.launch`
`$rosrun mavros mavsys mode -c OFFBOARD`
`$rosrun mavros mavsafety arm`

I want now jump into a real matter, e.g. build a `3D octree` map by means of a `global_planner` to be used later as a start point to plan the path by means of a `global_planner` and see it in `RVIZ`.

1.-I can get a ZED pointcloud2 topic `/stereo_camera/point_cloud/cloud_registered` 

2.-Also the ZED odom topic `/stereo_camera/odom`

The question is which launch file from the `global_planner` package shall i use?

My best guess is to use `global_planner_octomap.launch` if is it like this do I need to set up or take away certain parameters?

Thank you.

mrivi, 2019-05-02:
@acpl00 Are you asking for a launch file for a test on hardware? If so, there isn't any launch file that does that for you out of the box. 
`global_planner_octomap.launch` launches the global planner and the octomap server. In order to run the algorithm on a vehicle you also need to start from you launch file mavros, your camera driver, transformations from the camera frame to the fcu frame. You can look at an example from the local planner on how to put together these components (https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_aeroD415.launch#L8-L27). Furthermore the README hardware section on this package will provide more information about how to configure the paramters

acpl00, 2019-05-02:
Hi mrivi,

well, yea a launch file to test on hardware.

The only thing I want to do now  is to build up a `3D octree map` and save it, I guess for that I do not need the `mavros` node, right? neither the `global_planner`, right?

Later on, when I want to plan a path and make the drone follow it, I would like to use the saved map (as a starting map) together with the global_planner node. Because, I have noticed that  in the simulation of  `$roslaunch global planner_sitl_mavros.launch` it is difficult to set the point goal since there is no initial map.

mrivi, 2019-05-02:
@acpl00 You need the transformation from the reference local_origin frame to the fcu and then your camera. For that you need mavros running. 

You can use the planner in mission mode. This allows you to set goals as mission waypoints in a map in QGroundControl. This is much more user friendly. You can try it in simulation as well.

acpl00, 2019-05-02:
Hi, 

I have used the Qground control station and way points in a mission plan mode, however the main Idea is having a 3D octree map as an initial map, 

then 

choosing a goal pose base on the initial map the drone can travel to  the goal position base on the global_planner node. On the way to the goal the drone is updating the 3D map as it is shown in the simulations  based on the command `$roslaunch global planner_sitl_mavros.launch` and also the user must have control all the time over the drone.

My pixhawk for the moment is broken, another is on the way, I am thinking just use the `global_planner_octomap.launch` together with the camera driver without `mavros` to build up the 3D octree map.

Is that possible?

Thank you.

mrivi, 2019-05-02:
No, you need to know where your vehicle is to build a map. That information is coming from mavros.

acpl00, 2019-05-02:
@mrivi 

Yes the position is necessary but this can be used from the camera odom, right? The ZED delivers a odom information. and probably later on when I get the pixhawk I can used mavros, right?

acpl00, 2019-05-06:
Hi,

1.- I am a bit confuse, because I want to build up a 3D octree map of some outdoor environment and then save it and then integrate it  into the `global_planner`  as a start 3D octree map.  Is this possible? 

I am also wondering which launch file I may take as a basis to start modifying it into my needs. In some post was I told that I could use `local_planner_A7001cam.launch`  and the make the necessary changes:

`local_planner -> global_planner`
`realsense -> zed camera`

But the launch file is not longer at github.

3.- Do you recommend first simulate it in rviz-gazebo  before launching it.

Looking forward to your replay.

Thank you.

Jaeyoung-Lim, 2019-05-06:
@acpl00 I am not sure if I understand you correctly. - Are you trying to use the global planner? or the local planner?

You can check the global planner launchfiles in case you want to use the global planner. `global_planner_node` is not the equivalent of `local_planner_node`. You need more nodes to run the global planner

Saving the octomap is possible through the octomap_server_node. You can check on how to do it [here](http://wiki.ros.org/octomap_server)

acpl00, 2019-05-06:
Hi

I would like to use the global planner, however which global planner launch file I can start using?

nicovanduijn, 2019-05-06:
I'm pretty sure you would need to write your own

acpl00, 2019-05-06:
well, but there is any launch file that I can start with, I mean to have as a start point and then modify it.

nicovanduijn, 2019-05-06:
My best suggestion is to start with SITL.
`global_planner_sitl_mavros.launch` will be a good start.

Then, if all your stuff works well and reliably in simulation, you can start thinking about testing on a real drone. For this, start from the same launchfile and think about what's different. You probably won't need to start gazebo anymore, right? Maybe add your ZED camera launchfiles. Then make sure you have the proper setup to make MAVROS connect to your flight controller (UDP ports etc.)

But again, SITL first. Anything else is just plain wrong

acpl00, 2019-05-06:
OK, thank you so much :)

closed 2019-05-06 by nicovanduijn