PX4/PX4-Avoidance #337: Unifying behavioral differences between local planner and global planner

Jaeyoung-Lim, 2019-05-03:
**Describe problem solved by the proposed feature**
I have been looking into understand how global planner works, and it seems that the implementation is quite different compared to the local planner.

This is not too bad, but what the real problem is that the general interaction with the firmware is different and the recent failsafe features in `local_planner` are not implemented in `global_planner`. 

For example, `global_planner` will never send `MAV_STATE_CRITICAL` to the firmware.

Re-implementing the same behavior doesn't seem quite straight forward as both planners are tightly integrated into their use cases. Moreover, it would be hard to maintain both cmdloops to show the same behavior everytime there is something changing in the `local_planner`

**Describe your preferred solution**
One solution can be that there is an additional node that handles all interaction with mavros including the failsafe features, while global and local planners only have planner specific functions, which interact with this additional node. 

I would like to implement something like a `PathHandlerNode` which does the following things, on top of #334 

- Takes states/data from mavros and provides an interface for the planner to access it
- Takes the output of the planner and passes it on to mavros
- Do fail safe checks and general companion status handling
- Make the `PathHandlerNode` object a member of the planners, so it doesn't need to be launched as a separate node as done in the global planner

Please give any inputs or comments regarding this issue. I would be willing to implement this, but this would need to bring changes on both `global_planner` and `local_planner` cmdloops, wanted some consensus from the community if this is something we want, or worth considering, or if it is too early to move in this direction.

nicovanduijn, 2019-05-06:
Personally, I like the idea.

Modularizing these things would also make it easier to use things like OMPL, or quickly sketch out new planners and test them. Maybe a semi-global planner will be the future? (To be honest, that's what the current local planner pretty much is...)

Jaeyoung-Lim, 2019-05-07:
@nicovanduijn Awesome!

I agree. The direction I would like to see the avoidance repo go, is that anybody should be able to easily plug in their planner to the avoidance. This unloads people from tightly integrating their stuff to PX4. but this will be far in the future.