PX4/PX4-Avoidance #338: Flight

ChenChangHaoWS, 2019-05-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)) or from the Gazebo simulation (logs are located in `~/.ros/logs/`)
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
- Share a link to a rosbag showing the problem

**Additional context**
Add any other context about the problem here.

ChenChangHaoWS, 2019-05-05:
Add my rosgraph here:
[image]

mrivi, 2019-05-05:
I see I few issues in the output 
> [ WARN] [1557028161.770638191]: GP: No GPS fix

GPS lock is necessary to run a mission.

> FCU: ARMING DENIED: Avoidance system not ready

The avoidance isn't running properly most likely due to the pointcloud and/or local position not being available. See also this warning

> [ WARN] [1557028190.159888619]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.

You need to verify the GPS lock on your system. Can you run a mission without OA and see if it behaves properly?
Can you also check the pointcloud that the camera is producing and the rate at which it's published?

ChenChangHaoWS, 2019-05-06:
@mrivi 
Thank you so much for your reply!
Your advise about checking the pointcloud do help me find the problem. There is something wrong with tf. And after I rewrite the launch file, every thing goes fine now. 
So I will close the issue, hope it can help others.

closed 2019-05-06 by ChenChangHaoWS

DiegoHerrera1890, 2021-09-08:
Hello everyone.
I am facing the same issue using ros-gazebo
Ubuntu 18.04
Ros Melodic
intel realsense T265 with PX4 firmware

what do I need to modify?

deep-palette, 2022-06-25:
@Chris-cch 
hello i'm currently working on using px4-avoidance with zed2i camera
but i realize that when i make launch file using this "./tools/generate_launchfile.sh" 
there is a problem
can you please share some code to make launch file?