PX4/PX4-Avoidance #340: global_planner, RC transmitter, joystick

acpl00, 2019-05-07:
Dear people.

The  `$roslaunch global planner_sitl_mavros.launch` can successfully run on my laptop.

I have a question.

1.-Is it possible to fly the drone in the `global_planner` mode using either RC transmitter or joystick instead of given a goal point in RVIZ? I mean I would like to fly the drone in gazebo using either RC or joystick.

2.- In case it is possible, how can it be achieved ?

Thank you.

nicovanduijn, 2019-05-07:
Kind of.

So first of all, what's the point of the global planner when you're flying manually? The global planner will only steer the drone in _offboard_ or _mission_ mode. If you just want to switch between mission and manual flight - then of course you can do so with a joystick.

Maybe what you're looking for is collision prevention - a mode where you steer the drone manually but it keeps you from flying into obstacles (as long as they are in the field of view of the sensor, of course). This functionality built into the **local planner** though, not the global planner.

To steer the drone with RC I'm not sure. Maybe you can run HIL with avoidance? I've never tried.

But you can connect a joystick quite easily. How to do so is documented [here](https://docs.qgroundcontrol.com/en/SettingsView/VirtualJoystick.html)  for an on-screen virtual joystick and [here](https://docs.qgroundcontrol.com/en/SetupView/Joystick.html) for a USB joystick.

Hope that helps

acpl00, 2019-05-07:
Thank you for your answer.

well, I have a tbsd discovery drone with a zed camera, a pixhawk 4 mini and px4flow sensor.

And the main idea is:

1.-To simulate the drone in `global_planner` mode by running the command `$roslaunch global planner_sitl_mavros.launch`

But before that I would like:

2.- To fly the drone in gazebo manually by means of a RC transmitter,  if not by means of a joystick. The previous is done for the purpose of  building a up an initial 3D octree map than later I can use it as a start point before I fly the drone in autonomous mode by means of the command 
`$roslaunch global_planner_sitl_mavros.launch`

3.- When the command `$roslaunch global_planner_sitl_mavros.launch` is running I would like to have full control over the drone by means of the RC transmitter.

I would like to simulate first in Gazebo-Rviz then do it in real.

4.- I was told  that it can be done using the  `global_planner_sitl_mavros.launch` as a start point with a proper modification, however I have the impression that more emphasis is put to the `local_planner` by the team, right? 
I do not know if I have explained myself clear :)

nicovanduijn, 2019-05-07:
Correct, the development team is currently focused on the local planner.

You should be able to run the global planner as you suggested, fly around in position control with the joystick. This will populate the octomap, as can be seen in RViz. You can switch from position into mission mode any time you want, and it will go from there.

We haven't tested the global planner in real flight in a very long time, so I'd love to see your results!

As a precaution, please make sure everything works super well in simulation first.

acpl00, 2019-05-07:
Hi thank you again for your answer.

1.-To put the drone into position control with the joystick is just:
       * run the command `$roslaunch global_planner_sitl_mavros.launch`
       * Then, `rosrun mavros mavsys mode -c OFFBOARD`
       * Then, `rosrun mavros mavsafety arm`
       * And then, fly the drone with the joystick.

2.- To change from position into mission mode is:
      * To select a goal point in `RVIZ`, right?

3.- Do you think that the `local_planner` would be a better option to do the things I would like to implement?

Thank you.

nicovanduijn, 2019-05-07:
1) No. the second line enables the offboard mode, you can no longer use the joystick if you do so. run the roslaunch, then Open QGC to make sure to have the joystick connected. Then you can arm and mode-switch with the joystick. Use QGC's mission planner to define and upload a mission.
2) No. You can enable mission from QGC, or (depending on your setup) with the joystick. The goal clicking in RViz is for offboard mode. This is not what you want
3) Yes. Except if having an octomap is a hard constraint.

acpl00, 2019-05-07:
Hi, thank you again for your answer.

1.- I understood :)

2.- Mission planner mode in `QGCS` is to select way points, right? But I would like that the drone follow a trajectory path towards a goal point, and this can be achieved by selecting a goal point in` RVIZ`, right?

3.- Maybe I shall first run the `$roslaunch global_planner_sitl_mavros.launch` then try the `local_planner` to get an insight. 

4.- But with `local_planner ` goal points and autonomous flight can be achieved, can not be?

Thank you

mrivi, 2019-05-07:
Well in the end the behaviour is the same. The vehicle will follow a straight line between the current position and the goal if there aren't any obstacles. If there are it will plan around them. I would suggest you try both approaches (goal click in rviz and mission mode) in simulation. 
Both local and global planner support mission and offboard mode.

acpl00, 2019-05-07:
Yea I will try both, lets see how it goes :)

By the way, I have set up the joystick in `QGCS`

- I first run the command `$roslaunch global_planner_sitl_mavros.launch` 
- then, I try to fly the drone with the joystick, but it does not respond. 
- Just wondering whether it might be possible to see the `gazebo` map in `QGCS`. 

Thank you

mrivi, 2019-05-07:
Have you followed the guide on how to properly setup a joystick? What steps are you doing? Are you using a USB joystick?

No you can't.

acpl00, 2019-05-07:
It was the joystick, I just chaged it, no it is working.

- I first press a  bottom that has been enable as a position mode.

- Then, I press a bottom that has been enable as a arm mode.

- Then, I start flying the drone.

-I guess, it is possible to save the map with rosrun octomap_server octomap_saver, right?

-When the octree 3D map is saved, do you think it is possible to load it when I run 
    `$roslaunch global_planner_sitl_mavros.launch`

-Do you also think is possible to change the camera for the ZED one, or maybe   is better just to wait when I run it in real, to disable gazebo and enable the ZED ROS Wrapper.

Jaeyoung-Lim, 2019-05-07:
@acpl00 While saving a octomap is a simple task, flying in a loaded octomap is not that simple. You need to have a state estimate that is consistent with the maop.

Simply loading the map will not make this happen, therefore you it is more likely you will get a shifted map, which will make your drone crash. You still need perception while flying.

acpl00, 2019-05-07:
well maybe  might be possible to load the map when running for the first time

`$roslaunch global_planner_sitl_mavros.launch`

- Then, when choosing a goal point in `rviz` the drone will start flying towards the goal and updating/fusing the map with the loaded one.

Do you think is that possible.

Also

- Do you also think is possible to change the existing camera in gazebo  for the ZED one, or maybe is better just to wait when I run it in real, I mean to disable gazebo and enable the ZED ROS Wrapper.

nicovanduijn, 2019-05-07:
No, this will not be easily possible.

Like Jay explained, the map is in the local coordinate frame. This resets every time you restart the drone. Therefore, if you load an octomap, it will likely be wrong, and there's no way you'll know.

Also, this is not an issue. Discussions like this should go to https://discuss.px4.io/

closed 2019-05-07 by nicovanduijn