PX4/PX4-Avoidance #341: Add limitation for local_planner parameters

sunzj, 2019-05-07:
**Describe problem solved by the proposed feature**
After PR281 https://github.com/PX4/avoidance/pull/281, local planner use the parameter from PX4 firmware. 

However, there isn't limitation for these parameters, as i test, the MPC_XY_CRUISE get from firmware is 10, even i already set the speed to 1 in mission mode. The local planner use 10 as param_mpc_xy_cruise,thus cause avoid obstacle fail.

**Describe your preferred solution**
Check and add limitation for local_planner parameters, not just use firmware parameters.

jkflying, 2019-05-07:
Long term, we want the local_planner (and the global_planner) to be completely transparent when there are no obstacles. This means that it should fly at the speed given by the MPC_XY_CRUISE during mission mode. Only once there are obstacles does the planner slow down the vehicle, usually due to the planned path being outside of the FOV of the camera.

So, the correct solution is to reduce MPC_XY_CRUISE. Otherwise there is a different behavior when obstacle avoidance is running or not, even when there are not any obstacles. MPC_XY_CRUISE can be updated live during flight, for testing.

jiejiezhang, 2019-05-08:
@jkflying  From the PX4 Firmware code we can see:  The MPC_XY_CRUISE is Normal horizontal velocity in AUTO modes (includes Mission/RTL / hold / etc). But when we perform the mission task, we can modify the speed.

[image]

In obstacle avoidance, we can't judge the speed by just using this parameter（MPC_XY_CRUISE）.

nicovanduijn, 2019-05-08:
I've gotten confused with this speed set by the mission item before. Back then, the speed set in the mission item is not being used by PX4. I just tested in SITL, and this seems to no longer be the case, so I guess you are correct

sunzj, 2019-05-08:
So the speed set in mission mode isn't MPC_XY_CRUISE, that's why i set the speed to 1 in mission mode, but still get 10 from PX4 firmware.

jkflying, 2019-05-08:
This makes sense, yes. We need to check the speed of the mission item and use that. It will be a little bit complicated since we need to find which messages they are from, but I agree it should be done to make it consistent with flying non-avoidance missions.