PX4/PX4-Avoidance #351: global_planer path planning issues

acpl00, 2019-05-14:
Dear people,

I have created a `.dae` 3D map in `sketchup` and have included the 3D map  into  a new `.world` file that I have created and at the same time I have  integrated it  into the   global_planer_sitl_mavros.launch.

`<arg name="world_path" default="$(find global_planner)/../sim/worlds/tbsd_3D.world" />`

The 3D .dae map based on where I want to make some real global path planning tests.

I have integrated the map into .world file as:

    
`<include> `
    `<uri>model://tbsd</uri>`
           `<name>tbsd</name>`
           `<pose>10 15 0 0 0 0</pose>`
   `</include>`

When I insert the following commands:

`$ rosrun mavros mavsys mode -c OFFBOARD`
 `$ rosrun mavros mavsafety arm`

The drone takes off.

The map and the take off drone is showed in the following figure.

[image]

The issues are:

- When I click in `RVIZ` for a `2D Nav Goal` in the other side of the building the drone tries to go over the building and it stops to some height and do nothing.

- Since I do not see the initial  map in `RVIZ` and If I choose the goal  where and obstacle is, the drone plans the path and the it flies to the goal and  crashes into the obstacle.

- Also it detects obstacles on the floor where there is nothing

Is it possible to overcome these issues and make the drone to fine the path and follow it even is over a high building and do not crash into obstacles?

Thank you and looking forward to your advise.

mrivi, 2019-05-14:
- planning is possible only up to 25m
- is the obstacle present in the pointcloud?
- the algorithm isn't able to distinguish between ground and an obstacle. If you want to add this feature contributions are more than welcomed.

acpl00, 2019-05-14:
Hi :) 

- Well, the height of the building is maximum  16 meters. Is it  a way to check that the planning is up to 25 meters.

- Yea the obstacle is present in the point cloud.

- You mean that due to the lack of this feature, the algorithm will detect objects on the ground floor  even though they are not present, as it had happened to me, right?   

- Well, if the distinction between ground floor and obstacle is not implemented, I would probably give a try to add that feature, just could you give me a hint where and how to start :) 

- Another issue, the drone takes off up to more or less 3.0 meters and then when I chose a goal in `rviz` it will always fly at 3 meters height, right? 

Thank you

- Is there a way to choose goal points coordinates at different different heights?

mrivi, 2019-05-15:
- you can check the vehicle local position. Run `rostopic echo /mavros/local_position/pose`. Which paramter did you change?

- The ground is there and the camera picks it up. What I meant is that the algorithm considers every pointcoud as an obstacle that needs to be avoidance. It doesn't classifies a ground pointcloud as such so if you fly at low altitude and the ground is in the FOV the algorithm will try to avoid it.

acpl00, 2019-05-15:
-  I have changed the max attitude parameter, it was at 10m and I have changed it to 17m because the height of the building is 16m.   The issue is that when I choose a goal position in `RVIZ` and the path is planned and the drone is flying to the goal, some times it just stops in the middle of the path and do  nothing. 

- Just wondering whether  the distinction between ground floor and obstacle feature shall be added.

mrivi, 2019-05-15:
Can you write the exact name of the parameter you changed and where did you change it? 
There is very little I can help you with the description you gave. Can you check the actual vehicle position when the vehicle stops, what's the console output?

acpl00, 2019-05-15:
- The output of  `$rostopic echo /mavros/local_position/pose` is:

`frame_id: "local_origin"`
`pose: `
  `position: `
    `x: 1.40365564823`
    `y: -3.61729264259`
    `z: 5.5078163147`
 ` orientation: `
   ` x: -0.00817611110047`
    `y: 0.00294000692462`
   ` z: 0.7070956392`
   ` w: 0.707064649557`

- I have changed the `int max_altitude_ = 17;` in `global_planner.h` before was 10.

mrivi, 2019-05-15:
ok, what does the console say? Can you copy paste the output here?

acpl00, 2019-05-15:
`===== Half-way path ====== `
`Search              iter_time overest   num_iter  path_cost `
`NodeWithoutSmooth   11        2         1         0         No path found, search in 2D `
`3.25 		 2.00 		 8 		 29.13 	0.00 		 2.00 		 1  0.00 	2.00` 		 `2.00 		 2 		 1.43 	`

acpl00, 2019-05-15:
I just post the pictures of `Gazebo` and `RVIZ`. 

In the beginning it finds the path over the building  then when it starts flying towards the wall the path gets cut off and the drone just stops and the plan path disappears.  Maybe is a matter to change a parameter. 

[image]

[image]

acpl00, 2019-05-15:
- Just want to mention that when I have flatten the roof, in other words when I took the attic away, the path was planned over the building and the drone flew towards the `rviz` goal, as it can be seen in the picture. This is weird for me.

[image]

acpl00, 2019-05-15:
- Just a general question :)
  - What shall be the best way to use global_planner algorithm in a real simulation?.  I mean, Is  the best way to use it together with QgroundControlStation? or just RVIZ and the real drone?

  -  What is the safest way to use` global_planner` algorithm?

mrivi, 2019-05-16:
@acpl00 
It is possible that you have run into a bug. I would suggest you go trough the code and try to figure out where the problem is.
As for outdoors tests, I would first make sure that the simulation is working properly. Then bench test your hardware setup, eg. how does the pointcloud look like, at which rate are the setpoints published etc.  t
There are already safety features in place such that if the planner stops sending setpoints the Firmware switches to Loiter flight mode.
I suggest to use QgroundControl with the vehicle in mission mode.

Jaeyoung-Lim, 2019-05-31:
@acpl00 One thing that can cause the global_planner to stop, is when there are no more waypoints left, but the goal is occupied.

closed 2019-07-03 by jkflying

optimize0107, 2019-07-08:
How can I change the world?

I changed "simple_obastacle" of the 'global_planner_stereo.launch' to something else like "boxes1"
`<arg name="world_file_name"    default="simple_obstacle" />`

World looks like changed. When launch 'global_planner_stereo.launch', I can see that Rviz map 
is not match to octomap. I want to see the changed map in gazebo like you. But if I run
`make px4_sitl_default gazebo`
, gazebo stuck at "preparing your world...". The message does not change even if I keep waiting.

mrivi, 2019-07-08:
@optimize0107 please don't post on a closed issue with a question not related to the initial issue.