PX4/PX4-Avoidance #353: global_planner_sitl_mavros.launch not there

20chix, 2019-05-14:
roslaunch global_planner global_planner_sitl_mavros.launch

fail to run because launch file is not there, did you guys rename it?

Jaeyoung-Lim, 2019-05-14:
@20chix The launchfiles were renamed, so that global planner and local_planner share more of the common interfaces.

For the same example, you can use `global_planner_stereo.launch`

20chix, 2019-05-15:
Thanks, might submit a pr for readme

20chix, 2019-05-15:
Sorry to bother you again but, but the launch file global_planner_stereo.launch  doesn't launch gazebo. It only launch rviz with green background.

FYI my gazebo path are set

Jaeyoung-Lim, 2019-05-15:
@20chix Could you check if the path is consistent with the current Readme? The `GAZEBO_MODEL_PATH` has changed recently.

The gazebo is still launched, but it is only the gzclient that is not being launched to save computation

20chix, 2019-05-15:
That's the `GAZEBO_MODEL_PATH` path ~/mav_ws/src/avoidance/avoidance/sim/models, which correspond to the path on the README.
Is there a new path for `GAZEBO_MODEL_PATH` ?

Jaeyoung-Lim, 2019-05-15:
Okay that is correct.and you dont see the tree.and the drone? Can you share the log when you run the launchfile?

20chix, 2019-05-15:
I rolled back to an older version. I'll try again soon with latest version